moononournation/Arduino_GFX #520: Can't use any gfx commands in Arduino loop

BenthamSoftwareServices, 2024-08-18:
When I add any gfx-> type command into loop() I get an error saying gfx is undefined, but it works if I use it in setup(), what am I doing wrong?

moononournation, 2024-08-19:
you may start with hello world example first.

BenthamSoftwareServices, 2024-08-19:
Thank you very much the "hello world" works without problem.  Can you see why my code following has error message gfx not declared in this scope, when I add a gfx command to loop, otherwise the gfx commands in setup work OK?

#include <Arduino_GFX_Library.h>

#define PIN_CS  22 //configurable
#define PIN_DC  16 //configurable
#define PIN_RES 4  //configurable
#define PIN_SDA 23 //fixed
#define PIN_SCL 18 //fixed

void setup()
{
	Arduino_DataBus *bus = new Arduino_HWSPI(PIN_DC, PIN_CS); //create data bus class

	Arduino_GFX *gfx = new Arduino_GC9A01(bus, PIN_RES, 0 /* rotation */, true /* IPS */, 240 /* width */, 240 /* height */); 

	gfx->begin(); //initialise gfx class
	
	gfx->fillScreen(gfx->color565(37, 234, 61));
	delay(2000);

	gfx->fillScreen(WHITE);

	gfx->setCursor(15, 110);
	gfx->setTextSize(3, 3);
	gfx->setTextColor(BLACK);
	delay(2000);
	gfx->println("Hello from setup!");
	}

void loop()
{
	delay(2000);
        gfx->println("Hello from loop!");
	delay(1000);
}

closed 2024-08-29 by moononournation