moononournation/Arduino_GFX #529: 'VSPI' was not declared in this scope

eagl1, 2024-08-30:
Hi,

I don't know how this happened ? All my previous compilations on another PC were OK, didn't get this error at all with the same library version and the same ESP32 board package.

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, QSPI PSRAM, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Enabled"

In file included from D:\projects_electronics_programming\tft\arduino sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino:5:

Arduino_GFX_dev_device.h:10:114: error: 'VSPI' was not declared in this scope

 Arduino_DataBus *bus = new Arduino_ESP32SPI(2 /* DC */, 15 /* CS */, 14 /* SCK */, 13 /* MOSI */, 12 /* MISO */, VSPI /* spi_num */);

                                                                                                                  ^~~~

D:\projects_electronics_programming\tft\arduino sketches\GFX_Library_for_Arduino\Arduino_GFX_dev_device.h:10:114: note: suggested alternative: 'SPI'

 Arduino_DataBus *bus = new Arduino_ESP32SPI(2 /* DC */, 15 /* CS */, 14 /* SCK */, 13 /* MOSI */, 12 /* MISO */, VSPI /* spi_num */);

                                                                                                                  ^~~~

                                                                                                                  SPI

exit status 1

'VSPI' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

moononournation, 2024-08-30:
you can skip this parameter and use default

eagl1, 2024-09-14:
Hi,

According to the ESP32 pinout, the SPI pins @`Arduino_GFX_dev_device.h` line 125, are for the `HSPI`
 not `VSPI`, it's a small typo.

I wanted to fork the library but I think changing `VSPI` to `HSPI` will require me to change its source which I don't know where it came from.

closed 2024-09-18 by moononournation