moononournation/Arduino_GFX #530: Compilation error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32PAR8Q'?

fenixfz, 2024-08-30:
good afternoon. I try to compile the PDQgraphicstest sketch example project but it gives me this error.
I have already completely reinstalled the Arduino IDE and the libraries and package boards. and still gives this error. Everything is updated and I don't understand why this error occurs. If possible, could you please help me........

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

fenixfz, 2024-08-31:
good morning.
First of all, thank you for your help.
If I understand you, you are telling me to downgrade the version I am using.  I had already tried with the Arduino esp32 2.0.10 version but it gave me the error when compiling "ping timout".  But I will try other versions.  Do you know of any version that is more stable?

fenixfz, 2024-08-31:
> please use arduion-esp32 2.x

Good morning.
First of all, thank you for your help.
If I understand you, you are telling me to downgrade the version I am using. I had already tried with the Arduino esp32 2.0.10 version but it gave me the error when compiling "ping timout".  But I will try other versions.  Do you know of any version that is more stable?

fenixfz, 2024-09-01:
It didn't work... I tried with several versions of gfx library 1.0.1, 1.2.1, 1.29, 1.3.0, 1.3.5, 1.3.9, 1.4.7 and with several versions of the esp32 board 1.0.0, 2.0 1, 2.0.6, 2.0.7, 2.0.8, 2.0.17, 3.0.3, 3.0.4.  I tried with all the combinations but it doesn't work or gives the error " Compilation error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32PAR8Q'?"  or gives the error "ping timout".
Can you help me

Big-mouth-c, 2024-09-03:
please use ESP32S3 Dev Module

fenixfz, 2024-09-03:
Me with these definitions.  esp32s3 board, gfx library for arduino v1.4.1 and with the esp32 board version for espressif v2.0.13.  I managed to run some demos.

fenixfz, 2024-09-03:
I needed help with just one more thing.
I'm trying to implement a test I made in Squareline Studio but the UI isn't starting and the screen stays white. Maybe someone could help me, this is my code.

#include <lvgl.h>
#include <Arduino_GFX_Library.h>
#include "ui.h"
#include "Arduino.h"
#include <TAMC_GT911.h>

#define TOUCH_SCL 20
#define TOUCH_SDA 19
#define TOUCH_INT -1
#define TOUCH_RST 38
#define TOUCH_ROTATION ROTATION_NORMAL
#define TOUCH_MAP_X2 0
#define TOUCH_MAP_X1 800
#define TOUCH_MAP_Y2 480
#define TOUCH_MAP_Y1 0

//---- TFT Set ------------------------------------------
#define GFX_BL 2
// 7"
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

int touch_last_x = 0;
int touch_last_y = 0;

TAMC_GT911 ts = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, max(TOUCH_MAP_X1, TOUCH_MAP_X2), max(TOUCH_MAP_Y1, TOUCH_MAP_Y2));

//---- Funções de Inicialização --------------------------------------------------

void tft_init() {
    // Inicializa o display TFT
gfx->begin();
gfx->fillScreen(0xFFFF); // white
Serial.println("TFT initialized");

}

void touch_init() {
    // Inicializa o touch screen
    ts.begin(); // Inicializa o touch screen
    Serial.println("Touch initialized"); // Confirma a inicialização
}

//---- Main --------------------------------------------------

void setup() {
    Serial.begin(115200);

    pinMode(GFX_BL, OUTPUT);  // Define o pino de controle da luz de fundo como saída
    digitalWrite(GFX_BL, HIGH);
    delay(10); // Espera 10ms para a luz de fundo estabilizar

    tft_init();   // Inicializa o TFT
    lv_init();
    touch_init(); // Inicializa o touch screen

    // Chama a função para inicializar a interface gerada pelo SquareLine Studio
    ui_init();
   
    Serial.println("Setup done");
}

void loop() {
    lv_timer_handler(); /* Deixa o GUI fazer seu trabalho */
    delay(5);
}

[image]

fenixfz, 2024-09-03:
> please use ESP32S3 Dev Module

Me with these definitions. esp32s3 board, gfx library for arduino v1.4.1 and with the esp32 board version for espressif v2.0.13. I managed to run some demos.

//---- TFT Set ------------------------------------------
#define GFX_BL 2
// 7"
Arduino_ESP32RGBPanel rgbpanel = new Arduino_ESP32RGBPanel(
41 / DE /, 40 / VSYNC /, 39 / HSYNC /, 42 / PCLK /,
14 / R0 /, 21 / R1 /, 47 / R2 /, 48 / R3 /, 45 / R4 /,
9 / G0 /, 46 / G1 /, 3 / G2 /, 8 / G3 /, 16 / G4 /, 1 / G5 /,
15 / B0 /, 7 / B1 /, 6 / B2 /, 5 / B3 /, 4 / B4 /,
0 / hsync_polarity /, 180 / hsync_front_porch /, 30 / hsync_pulse_width /, 16 / hsync_back_porch /,
0 / vsync_polarity /, 12 / vsync_front_porch /, 13 / vsync_pulse_width /, 10 / vsync_back_porch */);

Arduino_RGB_Display gfx = new Arduino_RGB_Display(
800 / width /, 480 / height /, rgbpanel, 0 / rotation /, true / auto_flush */);

fenixfz, 2024-09-03:
> please use ESP32S3 Dev Module

I needed help with just one more thing.
I'm trying to implement a test I made in Squareline Studio but the UI isn't starting and the screen stays white. Maybe someone could help me, this is my code.

#include <lvgl.h>
#include <Arduino_GFX_Library.h>
#include "ui.h"
#include "Arduino.h"
#include <TAMC_GT911.h>

#define TOUCH_SCL 20
#define TOUCH_SDA 19
#define TOUCH_INT -1
#define TOUCH_RST 38
#define TOUCH_ROTATION ROTATION_NORMAL
#define TOUCH_MAP_X2 0
#define TOUCH_MAP_X1 800
#define TOUCH_MAP_Y2 480
#define TOUCH_MAP_Y1 0

//---- TFT Set ------------------------------------------
#define GFX_BL 2
// 7"
Arduino_ESP32RGBPanel rgbpanel = new Arduino_ESP32RGBPanel(
41 / DE /, 40 / VSYNC /, 39 / HSYNC /, 42 / PCLK /,
14 / R0 /, 21 / R1 /, 47 / R2 /, 48 / R3 /, 45 / R4 /,
9 / G0 /, 46 / G1 /, 3 / G2 /, 8 / G3 /, 16 / G4 /, 1 / G5 /,
15 / B0 /, 7 / B1 /, 6 / B2 /, 5 / B3 /, 4 / B4 /,
0 / hsync_polarity /, 180 / hsync_front_porch /, 30 / hsync_pulse_width /, 16 / hsync_back_porch /,
0 / vsync_polarity /, 12 / vsync_front_porch /, 13 / vsync_pulse_width /, 10 / vsync_back_porch */);

Arduino_RGB_Display gfx = new Arduino_RGB_Display(
800 / width /, 480 / height /, rgbpanel, 0 / rotation /, true / auto_flush */);

int touch_last_x = 0;
int touch_last_y = 0;

TAMC_GT911 ts = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, max(TOUCH_MAP_X1, TOUCH_MAP_X2), max(TOUCH_MAP_Y1, TOUCH_MAP_Y2));

//---- Funções de Inicialização --------------------------------------------------

void tft_init() {
// Inicializa o display TFT
gfx->begin();
gfx->fillScreen(0xFFFF); // white
Serial.println("TFT initialized");

}

void touch_init() {
// Inicializa o touch screen
ts.begin(); // Inicializa o touch screen
Serial.println("Touch initialized"); // Confirma a inicialização
}

//---- Main --------------------------------------------------

void setup() {
Serial.begin(115200);

pinMode(GFX_BL, OUTPUT);  // Define o pino de controle da luz de fundo como saída
digitalWrite(GFX_BL, HIGH);
delay(10); // Espera 10ms para a luz de fundo estabilizar

tft_init();   // Inicializa o TFT
lv_init();
touch_init(); // Inicializa o touch screen

// Chama a função para inicializar a interface gerada pelo SquareLine Studio
ui_init();

Serial.println("Setup done");
}

void loop() {
lv_timer_handler(); /* Deixa o GUI fazer seu trabalho */
delay(5);
}

[image]

moononournation, 2024-09-03:
Please use Arduino-esp32 v2.x

closed 2024-09-03 by moononournation