moononournation/Arduino_GFX #534: Compilation error : esp_memory_utils.h: No such file or directory

bfrmd, 2024-09-08:
Hi,

I’m using Arduino_GFX Library v 1.4.8 with an ESP32-8048S050C display board. It is very similar to the ESP32-8048S043C, except for the size of the LCD panel (same 800x480 definition, same ESP32-S3 WROOM1 chip, same ST7262 driver chip).
I’m trying to compile the example sketch PDQgraphicstest.ino.

 My configuration is as follows :
-	Arduino IDE v 2.3.2
-	ESP 32 Espressif System Board Package v 2.0.17
-	Board selected in the Arduino IDE: ESP32S3 Dev Module with Flash size 16 MB and OPI PSRAM

In the Arduino_GFX_dev_device.h folder, I’ve uncommented the line  #define ESP32_8048S043.

Instead, I’ve also tried to uncomment the line #define ESP32_4827S043 while commenting the corresponding option 1 and uncommenting option 2 (ST7262 IPS LCD 800x480).

Regardless of the uncommented lines in the Arduino_GFX_dev_device.h file, the compilation aborts with this error :

_C:\users\bf\Documents\Arduino\libraries\Arduino_GFX-master\src/databus/Arduino_ESP32SPIDMA.h:7:10: fatal error: esp_memory_utils.h: No such file or directory
 #include <esp_memory_utils.h>
^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Compilation error: exit status 1_

The compilation error is exactly the same on another Windows 10 PC with Arduino IDE 1.8.19 and the same configuration.

With ESP 32 Espressif System Board Package v 3.0.4, I don’t get this esp_memory_utils  error, but instead, I get the well-known "Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32PAR8Q'? I know that this package V3 is not compatible with the Arduino_GFX library…

How to solve the esp_memory_utils  issue?

Thanks

jfseaman, 2024-09-09:
I have the same issue in 1.4.8

ESP32 Boards def 2.0.17 because Adafruit Qualia ESP32S3 RGB666 needs Arduino_ESP32RGBPanel was lost in boards 3.0.x

jfseaman, 2024-09-09:
A quick test.

Comment out the line 7 and everything works fine under boards 2.0.17

bfrmd, 2024-09-09:
> A quick test.
> 
> Comment out the line 7 and everything works fine under boards 2.0.17

@jfseaman: Thanks for your advice. However, I'm not sure I understand. In which file should I comment out line 7 to eliminate the fatal compilation error mentioned in the title of this thread: "esp_memory_utils? h: No such file or directory"?

moononournation, 2024-09-09:
fix in v1.4.9