moononournation/Arduino_GFX #542: Rotation fails with version > 1.4.1

MisterMartin, 2024-10-01:
I use the rotation parameter (==1) in ```Arduino_ESP32RGBPanel()``` so that the rectangular panel may be
horizontally displayed. 

This works correctly with Arduino_GFX v1.4.1. For later versions, _only the left
third of the display is rendered_.

If no rotation is applied (==0), it renders properly in all versions >= v1.4.1.

Thanks for your help!

Processor: Adafruit Qualia ESP32-S3 for RGB-666
Display: Adafruit TL032FWV01CT 320x820
Toolchain: ArduinoIDE or PlatformIO
LVGL: v9.2.0
Arduino_GFX: v1.4.1, or v1.4.9

*v1.4.1 displays:*
[image]

*v1.4.9 displays:*
[image]

Here is the code, which is a trimmed down version of _examples/LVGL/LVGL_Arduino_V9/_
```cpp
#include <lvgl.h>
#include <Arduino_GFX_Library.h>
#define GFX_BL DF_GFX_BL

static Arduino_XCA9554SWSPI *expander = new Arduino_XCA9554SWSPI(PCA_TFT_RESET, PCA_TFT_CS, PCA_TFT_SCK, PCA_TFT_MOSI, &Wire, 0x3F);
static Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    TFT_DE, TFT_VSYNC, TFT_HSYNC, TFT_PCLK, TFT_R1, TFT_R2, TFT_R3, TFT_R4, TFT_R5, TFT_G0, TFT_G1, TFT_G2, TFT_G3, TFT_G4, TFT_G5, TFT_B1, TFT_B2, TFT_B3, TFT_B4, TFT_B5,
    1 /* hsync_polarity */, 50 /* hsync_front_porch */, 2 /* hsync_pulse_width */, 44 /* hsync_back_porch */,
    1 /* vsync_polarity */, 16 /* vsync_front_porch */, 2 /* vsync_pulse_width */, 18 /* vsync_back_porch */
                                                                                      //    ,1, 30000000
);
static Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    // 3.2" 320x820 rectangle bar display
    320 /* width */, 820 /* height */, rgbpanel, 1 /* rotation */, true /* auto_flush */,
    expander, GFX_NOT_DEFINED /* RST */, tl032fwv01_init_operations, sizeof(tl032fwv01_init_operations));

static lv_display_t *disp;
static lv_color_t *disp_draw_buf;

static uint32_t millis_cb(void)
{
    return millis();
}
static void my_disp_flush(lv_display_t *disp, const lv_area_t *area, uint8_t *px_map)
{
    uint32_t w = lv_area_get_width(area);
    uint32_t h = lv_area_get_height(area);
    gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)px_map, w, h);
    lv_disp_flush_ready(disp);
}
void setup()
{
    Serial.begin(115200);

    gfx->begin();
    gfx->fillScreen(BLACK);
    pinMode(GFX_BL, OUTPUT);
    digitalWrite(GFX_BL, HIGH);
    lv_init();
    lv_tick_set_cb(millis_cb);

    disp_draw_buf = (lv_color_t *)heap_caps_malloc(gfx->width() * 40 * 2, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
    disp = lv_display_create(gfx->width(), gfx->height());
    lv_display_set_flush_cb(disp, my_disp_flush);
    lv_display_set_buffers(disp, disp_draw_buf, NULL, gfx->width() * 40 * 2, LV_DISPLAY_RENDER_MODE_PARTIAL);

    lv_obj_t *label = lv_label_create(lv_scr_act());
    lv_label_set_text(label, "Hello Arduino, I'm LVGL!(V" GFX_STR(LVGL_VERSION_MAJOR) "." GFX_STR(LVGL_VERSION_MINOR) "." GFX_STR(LVGL_VERSION_PATCH) ")");
    lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
}
void loop()
{
    lv_task_handler();
    delay(5);
}
```

moononournation, 2024-10-04:
I think you should use `820 /* width */, 320 /* height */`

closed 2024-10-04 by moononournation

MisterMartin, 2024-10-04:
This has been tried. All combinations of width x height x rotation have been tried. It does not work.

sugartechllc, 2024-10-04:
Looking at the changes between v1.4.1 and 1.4.2 (https://github.com/moononournation/Arduino_GFX/compare/v1.4.1...v1.4.2),
in `Arduino_RGB_Display::writePixelPreclipped()`, the changes that reference the fb pointer look odd. 

starting at line 74, in a switch statement indexed  on `_rotation`, for case 1:
```cpp
    fb += (int32_t)x * _height;
    fb += _max_y - y;
```
is changed to:
```cpp
    fb += (int32_t)x * _fb_width;
    fb += _fb_max_x - y;
```

For case 3 we have:
```cpp
    fb += (int32_t)(_max_x - x) * _height;
```
is changed to:
```cpp
    fb += (int32_t)(_fb_max_y - x) * _fb_width;
```

These changes don't seem consistent with all of the other changes that are related to `_rotation` in 
`Arduino_RGB_Display`

Perhaps it is correct, but I don't see the frame buffer indexing being changed in the same way elsewhere
in the class.

MisterMartin, 2024-10-04:
Can this issue be reopened? Thanks!

moononournation, 2024-10-05:
Thanks for point this changes, it should be typo in draw16bitRGBBitmap() while this changes. Please get update to check is it fixed.

MisterMartin, 2024-10-05:
Testing with sha a0217f61.

Unfortunately that didn't fix it. While the display is now completely filled, it is just hash now.

Maybe usage of ROW_OFFSET and COL_OFFSET are rotation dependent?

I suspect that this problem will appear for any non-square display. Are you able to try it on one of those?

I also tried adding the fixes that I had mentioned a few posts back (with a0217f61), but that didn't help, probably because the frame buffer indexing is messed up elsewhere. 

Thanks!

[image]

moononournation, 2024-10-06:
Sorry, I am not using Arduino-esp32 2.x right now. Could you help to get update and test again?

MisterMartin, 2024-10-07:
I suspect you would find the same problem on any non-rectangular display. I’m on the road right now, but I should be able to test the recent commit in the next day or two. I really appreciate you looking at this!

MisterMartin, 2024-10-16:
Fixed as of 4f9ca9bbb04ef51bb11789455e72dbfe5621c091. 

Thanks for sorting this out!