moononournation/Arduino_GFX #549: XIAO ESP-32C3

DLebedev1806, 2024-10-11:
Hello. Please help me. After the update, the following error appeared for ESP-C3

\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32SPI.cpp: In member function 'virtual bool Arduino_ESP32SPI::begin(int32_t, int8_t)':
\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32SPI.cpp:280:19: error: 'PERIPH_SPI2_MODULE' was not declared in this scope
  280 |   periph_ll_reset(PERIPH_SPI2_MODULE);
      |                   ^~~~~~~~~~~~~~~~~~
\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32SPI.cpp:280:3: error: 'periph_ll_reset' was not declared in this scope
  280 |   periph_ll_reset(PERIPH_SPI2_MODULE);
      |   ^~~~~~~~~~~~~~~
\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32SPI.cpp:281:3: error: 'periph_ll_enable_clk_clear_rst' was not declared in this scope
  281 |   periph_ll_enable_clk_clear_rst(PERIPH_SPI2_MODULE);

There is such an error for ESP-32C6

\Documents\Arduino\77777\77777.ino:47:28: error: expected type-specifier before 'Arduino_ESP32SPI'
   47 | Arduino_DataBus* bus = new Arduino_ESP32SPI(22 /* DC */, 20 /* CS */, 19 /* SCK */, 18 /* MOSI */, GFX_NOT_DEFINED /* MISO */);

moononournation, 2024-10-11:
it should fixed in master branch, you may try this out

DLebedev1806, 2024-10-11:
I tried from the master branch.

moononournation, 2024-10-12:
should fixed, please get update and test again

closed 2024-10-12 by moononournation