moononournation/Arduino_GFX #550: ESP32-C6

DLebedev1806, 2024-10-14:
Good afternoon. I'm sorry, but with ESP - 32C6, the project is still not going to work. I have attached a picture with an example project and an error. Thank you!
[image]

DLebedev1806, 2024-10-14:
#include <Arduino_GFX_Library.h>
Arduino_DataBus* bus = new Arduino_HWSPI(22 /* DC */, 20 /* CS */, 19 /* SCK */, 18 /* MOSI */, GFX_NOT_DEFINED /* MISO */);
Arduino_GFX* gfx = new Arduino_ST7789(bus, 21 /* RST */, 0 /* rotation */, true /* IPS */, 240 /* width */, 320 /* height */, 0 /* col offset 1 */, 0 /* row offset 1 */, 0 /* col offset 2 */, 80 /* row offset 2 */);

void setup() {
  if (!gfx->begin()) {  //Инициализация дисплея
  Serial.println("Display initialization failed!");
  }
}

void loop() {
  gfx->fillScreen(RED);
  gfx->setTextColor(WHITE);
  gfx->setTextSize(2);
  gfx->setCursor(0, 50);
  gfx->println("Test");
  delay(2000);
}

moononournation, 2024-10-14:
please use HWSPI

DLebedev1806, 2024-10-14:
Well, everything works with HWSPI, thanks!

closed 2024-10-14 by DLebedev1806