moononournation/Arduino_GFX #551: JC3248W535 device, (ESP32-s3 with 480x320 touch lcd) not working when used directly (without canvas)

danielezing, 2024-10-14:
Hi, I tested this library on a esp32s3 board with touch lcd (namely the JC3248W535 device, 480x320 touch lcd, I have several boards).
The library works OK (I tried the PDQgraphicstest ) when using the CANVAS, but if I try to use directly the dispolay (without canvas, I can't use so much RAM for frame buffer) it's not working at all.
In particular:
- no text output at all
- only some of the tests display something ... the fill screen is ok, the filled rectangles show some distorted ones ... but no lines, arcs, triangles ....

Is this library intended to be used only with a canvas, or can I use it directly on the SPI connected display?
I tried to slow down the SPI but no change.
Thank you!

danielezing, 2024-10-14:
Thank you for the quick reply .... 
Now my problem remains ... I can't afford a 480x320 frame buffer on an esp32 ....
Is there problem related to testing, or are some low level functions missing ?
As canvas is working (that is, when a complete image is copied in one block to the display) .... taking a look at "flush()" I see that actually there is a copy of the whole canvas in a pixel by pixel way ... so I can't figure out where could be the problem ...

moononournation, 2024-10-14:
this dev board have PSRAM, please enable it

closed 2024-10-15 by moononournation