moononournation/Arduino_GFX #555: Waveshare 2.1 esp32-s3 display support

themagicm, 2024-10-19:
Been messing with this trying to get it to work.

I get the backlight to turn on. LOL.  But stuck with just getting hello world.

Thank you!

themagicm, 2024-10-19:
What I defined:

`Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 39 /* VSYNC */, 38 /* HSYNC */, 41 /* PCLK */, /* RGB control pins */
    5  /* B1 */, 45 /* B2 */, 48 /* B3 */, 47 /* B4 */, 21 /* B5 */, /* Blue pins */
    14 /* G0 */, 13 /* G1 */, 12 /* G2 */, 11 /* G3 */, 10 /* G4 */, 9  /* G5 */, /* Green pins */
    46 /* R1 */, 3 /* R2 */, 8 /* R3 */, 18 /* R4 */, 17 /* R5 */, /* Red pins */
    1 /* hsync_polarity */, 50 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 10 /* hsync_back_porch */, /* Horizontal timings */
    1 /* vsync_polarity */, 8 /* vsync_front_porch */, 3 /* vsync_pulse_width */, 8 /* vsync_back_porch */, /* Vertical timings */
    1 /* pclk_active_neg */
);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */
);

`

moononournation, 2024-10-19:
The tested dev devices are listed in README.md. I cannot make a recommendation to you about untested dev devices and you should not raise an issue about this. Please stick to the source code manufacturer provided.

themagicm, 2024-10-19:
oh

Sent with [Proton Mail](https://proton.me/mail/home) secure email.

closed 2024-10-22 by moononournation