moononournation/Arduino_GFX #556: How to print refresh frames ?

eagl1, 2024-10-19:
Hi,

I'm trying to print for example, a simple counter and flush the display but it instead print sequentially.

This is the code:

```
void setup()
{
  Serial.begin(115200);

  // Init Display
  if (!gfx->begin()) { Serial.println("gfx->begin() failed!"); }
  else{Serial.println("gfx->begin() succeeded!");}

  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);

  gfx->fillScreen(BLACK);
  gfx->setCursor(0, 0);
  gfx->setTextSize(4);
  gfx->setTextColor(WHITE);
  gfx->println(F("Arduino GFX TFT_eSPI GT911 TEST"));
  gfx->println(F("Hello World!\n"));
  gfx->println(123);
}

void loop(void)
{
  static int i = 0;
  gfx->flush();
  gfx->print(i);
  i++;
  delay(500);
}
```

eagl1, 2024-10-19:
I've done this code and it's working:

```
#include "Arduino_GFX_TFT_eSPI.h"
#include "Arduino_GFX_dev_device.h"
#include "TAMC_GT911.h"

#define TOUCH_SDA    42//19
#define TOUCH_SCL    41//20
#define TOUCH_INT    -1//39
#define TOUCH_RST    40//38
#define TOUCH_WIDTH  480
#define TOUCH_HEIGHT 320

#ifdef ESP32
#undef F
#define F(s) (s)
#endif

int32_t w, h, n, n1, cx, cy, cx1, cy1, cn, cn1;
uint8_t tsa, tsb, tsc, ds;

TAMC_GT911 tp = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, TOUCH_WIDTH, TOUCH_HEIGHT);

void setup(){
  Serial.begin(115200);

  // Init Display
  if (!gfx->begin()){
    Serial.println("gfx->begin() failed!");
  }
  else{Serial.println("gfx->begin() succeeded!");}

  // initializing touch module
  tp.begin(GT911_ADDR1);
  tp.setRotation(ROTATION_NORMAL);

  w = gfx->width();
  h = gfx->height();
  n = min(w, h);
  n1 = n - 1;
  cx = w / 2;
  cy = h / 2;
  cx1 = cx - 1;
  cy1 = cy - 1;
  cn = min(cx1, cy1);
  cn1 = cn - 1;
  tsa = ((w <= 176) || (h <= 160)) ? 1 : (((w <= 240) || (h <= 240)) ? 2 : 3); // text size A
  tsb = ((w <= 272) || (h <= 220)) ? 1 : 2;                                    // text size B
  tsc = ((w <= 220) || (h <= 220)) ? 1 : 2;                                    // text size C
  ds = (w <= 160) ? 9 : 12;                                                    // digit size

  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);

  gfx->println(F("Benchmark\tmicro-secs"));
  delay(1000);
  gfx->fillScreen(BLACK);
  testFillScreen();
  gfx->fillScreen(BLACK);
  testText();
  gfx->fillScreen(BLACK);
  testPixels();
  gfx->fillScreen(BLACK);
  testLines();
  gfx->fillScreen(BLACK);
  gfx->println(F("Done!"));
  delay(1000);
  gfx->fillScreen(BLACK);  
}

void loop(void){
  
  tp.read();
  
  if (tp.isTouched){

    for (int i=0; i<tp.touches; i++){
      gfx->print(F("Touch "));  gfx->println(i+1);
      
      gfx->print(F(" ")); gfx->println((uint16_t)tp.points[i].x);
      gfx->print(F(" ")); gfx->println((uint16_t)tp.points[i].y);
      gfx->println();
    }
  }
  delay(500);
  gfx->setCursor(0, 0);
  gfx->flush();
  gfx->fillScreen(BLACK);  
}

void testFillScreen(){
  // Shortened this tedious test!
  gfx->fillScreen(WHITE);
  gfx->fillScreen(RED);
  gfx->fillScreen(GREEN);
  gfx->fillScreen(BLUE);
  gfx->fillScreen(BLACK);
}

void testText(){
  gfx->setCursor(0, 0);

  gfx->setTextSize(1);
  gfx->setTextColor(WHITE, BLACK);
  gfx->println(F("Hello World!"));

  gfx->setTextSize(2);
  gfx->setTextColor(gfx->color565(0xff, 0x00, 0x00));
  gfx->print(F("RED "));
  gfx->setTextColor(gfx->color565(0x00, 0xff, 0x00));
  gfx->print(F("GREEN "));
  gfx->setTextColor(gfx->color565(0x00, 0x00, 0xff));
  gfx->println(F("BLUE"));

  gfx->setTextSize(tsa);
  gfx->setTextColor(YELLOW);
  gfx->println(1234.56);

  gfx->setTextColor(CYAN);
  gfx->println(F("hi,"));

  gfx->setTextSize(tsc);
  gfx->setTextColor(MAGENTA);
  gfx->println(F("hello,"));

  gfx->setTextSize(2);
  gfx->setTextColor(RED);
  gfx->println(F("Size 2"));

  gfx->setTextSize(3);
  gfx->setTextColor(ORANGE);
  gfx->println(F("Size 3"));

  gfx->setTextSize(4);
  gfx->setTextColor(YELLOW);
  gfx->println(F("Size 4"));

  gfx->setTextSize(5);
  gfx->setTextColor(GREENYELLOW);
  gfx->println(F("Size 5"));

}

void testPixels(){

  for (int16_t y = 0; y < h; y++){
    for (int16_t x = 0; x < w; x++){
      gfx->drawPixel(x, y, gfx->color565(x << 3, y << 3, x * y));
    }
  }

}

void testLines(){
  int32_t x1, y1, x2, y2;

  x1 = y1 = 0;
  y2 = h - 1;
  for (x2 = 0; x2 < w; x2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }

  x2 = w - 1;
  for (y2 = 0; y2 < h; y2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }

  x1 = w - 1;
  y1 = 0;
  y2 = h - 1;
  for (x2 = 0; x2 < w; x2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }

  x2 = 0;
  for (y2 = 0; y2 < h; y2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }

  x1 = 0;
  y1 = h - 1;
  y2 = 0;
  for (x2 = 0; x2 < w; x2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }

  x2 = w - 1;
  for (y2 = 0; y2 < h; y2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }

  x1 = w - 1;
  y1 = h - 1;
  y2 = 0;
  for (x2 = 0; x2 < w; x2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }

  x2 = 0;
  for (y2 = 0; y2 < h; y2 += 6){
    gfx->drawLine(x1, y1, x2, y2, BLUE);
  }
}

```

eagl1, 2024-10-19:
What is the difference between

```
gfx->flush();
```
and
```
gfx->fillScreen(BLACK);
```

moononournation, 2024-10-22:
This example show fps at top left:
https://github.com/moononournation/Arduino_GFX/blob/master/examples/Sprite/SpriteGif/SpriteGif.ino

Hope it can help you.

closed 2024-10-22 by moononournation

eagl1, 2024-10-22:
I did this one.

```
void simple_counter(void){
  static int i = 0;
  gfx->setCursor(0, 0);
  gfx->fillScreen(BLACK);
  gfx->print(i);
  i++;
  delay(100);
}
```