moononournation/Arduino_GFX #569: error: narrowing conversion

umeiko, 2024-11-23:
when i compile this library as components of esp-idf with arduino, the error shown:
```shell
E:/idf_ws/host-uvc-esp32s3/components/Arduino_GFX/databus/Arduino_ESP32RGBPanel.cpp: In member function 'uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)':
E:/idf_ws/host-uvc-esp32s3/components/Arduino_GFX/databus/Arduino_ESP32RGBPanel.cpp:53:57: error: narrowing conversion of '((((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_prefer_speed == -1) ? ((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_speed : ((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_prefer_speed)' from 'int32_t' {aka 'int'} to 'unsigned int' inside { } [-Werror=narrowing]
           .pclk_hz = (_prefer_speed == GFX_NOT_DEFINED) ? _speed : _prefer_speed,
                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
```
seems like a error type conversion bug.

notascooby, 2024-12-18:
As per umeiko, I also get the same narrowing conversion warning 
Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:53:57: warning: narrowing conversion of '((((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_prefer_speed == -1) ? ((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_speed : ((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_prefer_speed)' from 'int32_t' {aka 'long int'} to 'uint32_t' {aka 'long unsigned int'} [-Wnarrowing]
   53 |           .pclk_hz = (_prefer_speed == GFX_NOT_DEFINED) ? _speed : _prefer_speed,

plus also:

Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:54:20: warning: narrowing conversion of 'w' from 'int16_t' {aka 'short int'} to 'uint32_t' {aka 'long unsigned int'} [-Wnarrowing]
   54 |           .h_res = w,
      |                    ^
Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:55:20: warning: narrowing conversion of 'h' from 'int16_t' {aka 'short int'} to 'uint32_t' {aka 'long unsigned int'} [-Wnarrowing]
   55 |           .v_res = h,
      |                    ^
Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:63:56: warning: narrowing conversion of '((((int)((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_hsync_polarity) == 0) ? 1 : 0)' from 'int' to 'uint32_t' {aka 'long unsigned int'} [-Wnarrowing]
   63 |               .hsync_idle_low = (_hsync_polarity == 0) ? 1 : 0,
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:64:56: warning: narrowing conversion of '((((int)((Arduino_ESP32RGBPanel*)this)->Arduino_ESP32RGBPanel::_vsync_polarity) == 0) ? 1 : 0)' from 'int' to 'uint32_t' {aka 'long unsigned int'} [-Wnarrowing]
   64 |               .vsync_idle_low = (_vsync_polarity == 0) ? 1 : 0,
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~

These did not occur prior to version 1.50.
Using Arduino IDE ver 1.8.19 with esp32 ver 3.1.0
Target ESP32S3

moononournation, 2024-12-23:
done

closed 2024-12-23 by moononournation