moononournation/Arduino_GFX #572: Touch problem with  JC3248W535 GUITION TFT

elektromos, 2024-11-27:
Hello!
I am using this configuration: #define GFX_DEV_DEVICE JC3248W535
#define GFX_BL 1

Arduino_DataBus *bus = new Arduino_ESP32QSPI(
45 /* CS */, 47 /* SCK */, 21 /* D0 */, 48 /* D1 */, 40 /* D2 */, 39 /* D3 */);
Arduino_GFX *g = new Arduino_AXS15231B(bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */, false /* IPS */, 320 /* width */, 480 /* height */);
#define CANVAS
Arduino_Canvas *gfx = new Arduino_Canvas(320 /* width */, 480 /* height */, g, 0 /* output_x */, 0 /* output_y */, 0 /* rotation */);

The display works fine, but the Touch does not work. Please, if anyone has an example of a touch pattern, please  help me!

closed 2024-12-02 by elektromos

NorthernMan54, 2024-12-05:
@elektromos Where did you set that config to enable touch to work ?

elektromos, 2024-12-05:
Very complicated