moononournation/Arduino_GFX #574: Error colors

vsalas89, 2024-12-01:
Hi,

I'm using a ESP32-S3-WROOM-u1 with SRAM octal. The display is ZZW395WHS (manf. zoneway) with ST7701 and GT911 panel driver. All is correct excepting the colors. I have made a test with version Arduino_GFX v1.2.1 and colors were right, but with Arduino_GFX v1.5.0 i get incorrect colors. I attach my code and a photo of the colors, anyone can help me?

```
#include <Arduino_GFX_Library.h>
#include "LibreFranklin_Light22pt7b.h"
#include "Roboto_Light22pt7b.h"

Arduino_DataBus *bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, 48 /* CS */,
    38 /* SCK */, 45 /* MOSI */, GFX_NOT_DEFINED /* MISO */
);

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    39 /* DE */, 42 /* VSYNC */, 41 /* HSYNC */, 40 /* PCLK */,
    7 /* R0 */, 15 /* R1 */, 16 /* R2 */, 17 /* R3 */, 18 /* R4 */,
    8 /* G0 */, 19 /* G1 */, 20 /* G2 */, 3 /* G3 */, 46 /* G4 */, 9 /* G5 */,
    10 /* B0 */, 11 /* B1 */, 14 /* B2 */, 21 /* B3 */, 47 /* B4 */,
    1 /* hsync_polarity */, 80 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 80 /* hsync_back_porch */,
    1 /* vsync_polarity */, 18 /* vsync_front_porch */, 2 /* vsync_pulse_width */, 30 /* vsync_back_porch */
);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, st7701_type1_init_operations, sizeof(st7701_type1_init_operations)
);

/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
  Serial.begin(115200);
  // Serial.setDebugOutput(true);
  // while(!Serial);
  Serial.println("Arduino_GFX Hello World example");

  // Init Display
  if (!gfx->begin())
  {
    Serial.println("gfx->begin() failed!");
  }
  gfx->fillScreen(0x0000);

  gfx->setCursor(24, 50);
  gfx->setFont(&LibreFranklin_Light22pt7b);
  gfx->setTextColor(BLUE); //0x07FF blue
  gfx->println("Blue");

  gfx->setCursor(24, 90);
  gfx->setFont(&Roboto_Light22pt7b);
  gfx->setTextColor(GREEN); //0xF81F green
  gfx->println("Green");

  gfx->setCursor(24, 130);
  gfx->setFont(&Roboto_Light22pt7b);
  gfx->setTextColor(RED); //0xFFC0 red
  gfx->println("Red");

  gfx->setCursor(24, 170);
  gfx->setFont(&Roboto_Light22pt7b);
  gfx->setTextColor(BLACK); //0xFFFF black
  gfx->println("Black");

  Serial.println(BLACK);

  delay(5000); // 5 seconds
}

void loop()
{
  delay(1000); // 1 second
}
```

Photo: [image]

vsalas89, 2024-12-01:
**FIXED!!!!**

I forgot to implement the init setup send it by the manufacturer, is not the same one as in the default Arduino_RGB_Display.h (st7701_type1_...). Here is the correct one:

```
static const uint8_t st7701_type1_init_operations[] = {
    BEGIN_WRITE,
    WRITE_COMMAND_8, 0xFF,
    WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x10,

    WRITE_C8_D16, 0xC0, 0x3B, 0x00,
    WRITE_C8_D16, 0xC1, 0x0D, 0x02,
    WRITE_C8_D16, 0xC2, 0x21, 0x08,

    WRITE_COMMAND_8, 0xB0, // Positive Voltage Gamma Control
    WRITE_BYTES, 16,
    0x00, 0x11, 0x18, 0x0E,
    0x11, 0x06, 0x07, 0x08,
    0x07, 0x22, 0x04, 0x12,
    0x0F, 0xAA, 0x31, 0x18,

    WRITE_COMMAND_8, 0xB1, // Negative Voltage Gamma Control
    WRITE_BYTES, 16,
    0x00, 0x11, 0x19, 0x0E,
    0x12, 0x07, 0x08, 0x08,
    0x08, 0x22, 0x04, 0x11,
    0x11, 0xA9, 0x32, 0x18,

    // PAGE1
    WRITE_COMMAND_8, 0xFF,
    WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x11,

    WRITE_C8_D8, 0xB0, 0x60, // Vop=4.7375v
    WRITE_C8_D8, 0xB1, 0x30, // VCOM=32
    WRITE_C8_D8, 0xB2, 0x87, // VGH=15v
    WRITE_C8_D8, 0xB3, 0x80,
    WRITE_C8_D8, 0xB5, 0x49, // VGL=-10.17v
    WRITE_C8_D8, 0xB7, 0x85,
    WRITE_C8_D8, 0xB8, 0x21, // AVDD=6.6 & AVCL=-4.6
    WRITE_C8_D8, 0xC1, 0x78,
    WRITE_C8_D8, 0xC2, 0x78,

    END_WRITE,

    DELAY, 20,

    BEGIN_WRITE,
    WRITE_COMMAND_8, 0xE0,
    WRITE_BYTES, 3, 0x00, 0x1B, 0x02,

    WRITE_COMMAND_8, 0xE1,
    WRITE_BYTES, 11,
    0x08, 0xA0, 0x00, 0x00,
    0x07, 0xA0, 0x00, 0x00,
    0x00, 0x44, 0x44,

    WRITE_COMMAND_8, 0xE2,
    WRITE_BYTES, 12,
    0x11, 0x11, 0x44, 0x44,
    0xED, 0xA0, 0x00, 0x00,
    0xEC, 0xA0, 0x00, 0x00,

    WRITE_COMMAND_8, 0xE3,
    WRITE_BYTES, 4, 0x00, 0x00, 0x11, 0x11,

    WRITE_C8_D16, 0xE4, 0x44, 0x44,

    WRITE_COMMAND_8, 0xE5,
    WRITE_BYTES, 16,
    0x0A, 0xE9, 0xD8, 0xA0,
    0x0C, 0xEB, 0xD8, 0xA0,
    0x0E, 0xED, 0xD8, 0xA0,
    0x10, 0xEF, 0xD8, 0xA0,

    WRITE_COMMAND_8, 0xE6,
    WRITE_BYTES, 4, 0x00, 0x00, 0x11, 0x11,

    WRITE_C8_D16, 0xE7, 0x44, 0x44,

    WRITE_COMMAND_8, 0xE8,
    WRITE_BYTES, 16,
    0x09, 0xE8, 0xD8, 0xA0,
    0x0B, 0xEA, 0xD8, 0xA0,
    0x0D, 0xEC, 0xD8, 0xA0,
    0x0F, 0xEE, 0xD8, 0xA0,

    WRITE_COMMAND_8, 0xEB,
    WRITE_BYTES, 7,
    0x02, 0x00, 0xE4, 0xE4,
    0x88, 0x00, 0x40,

    WRITE_C8_D16, 0xEC, 0x3C, 0x00,

    WRITE_COMMAND_8, 0xED,
    WRITE_BYTES, 16,
    0xAB, 0x89, 0x76, 0x54,
    0x02, 0xFF, 0xFF, 0xFF,
    0xFF, 0xFF, 0xFF, 0x20,
    0x45, 0x67, 0x98, 0xBA,

    //-----------VAP & VAN---------------
    WRITE_COMMAND_8, 0xFF,
    WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x00,

    WRITE_C8_D8, 0x36, 0x8, 

    WRITE_C8_D8, 0x3A, 0x66, // 0x70 RGB888, 0x60 RGB666, 0x50 RGB565

    WRITE_COMMAND_8, 0x11, // Sleep Out
    
    END_WRITE,

    DELAY, 200,

    BEGIN_WRITE,
    WRITE_COMMAND_8, 0x29, // Display On
    DELAY, 50,
    END_WRITE
};
```

moononournation, 2024-12-14:
wild guess it is not st7701_type1_init_operations, you may try other types...

closed 2024-12-14 by moononournation