moononournation/Arduino_GFX #576: 76x284 & 128x160 ST7735, RED and BLUE swapped

Marooned-MB, 2024-12-05:
I have two displays: 76x284 and 128x160.

76x284 is by default defined using Arduino_ST7789 but that just gave me the white screen with no image, so I've switched to ST7735.
All seems to be working except red and blue colors looks swapped on both displays.

Here is my setup for 76x284:
```
Arduino_DataBus *bus = new Arduino_ESP8266SPI(4 /* DC */, 15 /* CS */);
Arduino_GFX *gfx = new Arduino_ST7735(bus, 2 /* TFT_RST */, 3 /* rotation */, false /* IPS */, 76 /* width */, 284 /* height */, 82 /* col offset 1 */, 18 /* row offset 1 */, 82 /* col offset 2 */, 18 /* row offset 2 */);
```
and here for 128x160:
```
Arduino_GFX *gfx = new Arduino_ST7735(bus, 2 /* TFT_RST */, 0 /* rotation */);
```

I've tried 4 configurations: with IPS set to true and false and with invertDisplay true and false.
Here are the results for 75x284 but it looks exactly the same (color wise) for 128x160:
|                     | IPS=true | IPS=false |
| ------------- | --------- | ---------- |
| invert=true  | [image] | [image] |
| invert=false | [image] | [image] |

Is there a way to have proper colors on those?
Thanks

moononournation, 2024-12-05:
It should be ST7789 but not ST7735

Marooned-MB, 2024-12-05:
As already mentioned in the 2nd line of the first post, using ST7789 gives just white screen with no data.
```
Arduino_GFX *gfx = new Arduino_ST7789(bus, 2 /* TFT_RST */, 3 /* rotation */, false /* IPS */, 76 /* width */, 284 /* height */, 82 /* col offset 1 */, 18 /* row offset 1 */, 82 /* col offset 2 */, 18 /* row offset 2 */);
```

moononournation, 2024-12-05:
76x284 should be ST7789, you cannot expect right display with incorrect driver. If it show white screen, try lower the speed.

Marooned-MB, 2024-12-05:
Ok, so back to 128x160 for a while. It should use ST7735, right? That's the default driver in this lib, so:
```
Arduino_GFX *gfx = new Arduino_ST7735(bus, 2 /* TFT_RST */, 0 /* rotation */);
```
is this correct?
I've called different speeds, even down to
```
gfx->begin(500000)
```
Here is the effect (normal and inverted):

https://github.com/user-attachments/assets/2455d4b4-c51b-47fb-9a0b-f7032e6259ad

moononournation, 2024-12-05:
There are many parameters combinations for ST7735. No quick method, you need try each in PDQgraphictest one by one.

closed 2024-12-11 by moononournation

Marooned-MB, 2024-12-11:
Well, this issue is closed but is not resolved. Looks like there is some problem with switched red and blue data but that library does not support that I guess.