moononournation/Arduino_GFX #583: "Arduino_ESP32RGBPanel" support in ESP32-S3(WROOM1 N8R8/N16R8) and ST7701S lcd driver

Yash333code, 2024-12-18:
I want to run ST7701S driver supported LCD Display - HD371001C40, I am using custom board with ESP32-S3-WROOM1(N16R8).
Board support package "https://dl.espressif.com/dl/package_esp32_index.json" selected board in arduino IDE - "ESP32S3 Dev Module" and "Fri3d Badge 2024(ESP32-S3-WROOM-1)" code is
`#include <SPI.h>
#include <Arduino_GFX_Library.h>
#include <Adafruit_GFX.h>
#define GFX_BL 16 // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
  35 /* CS */, 38 /* SCK */, 39 /* SDA */
  33 /* DE */, 32 /* VSYNC */, 31 /* HSYNC */, 34 /* PCLK */,
  21 /* R0 */, 20 /* R1 */, 19 /* R2 */, 18 /* R3 */, 17 /* R4 */,
  9 /* G0 */, 12 /* G1 */, 25 /* G2 */, 24 /* G3 */, 23 /* G4 */, 22 /* G5 */,
  4 /* B0 */, 5 /* B1 */, 6 /* B2 */, 7 /* B3 */, 8 /* B4 */
  
);

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
  bus, GFX_NOT_DEFINED, 800 /* width */, 480 /* height */);
#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
    gfx->begin();
    gfx->fillScreen(BLACK);

#ifdef GFX_BL
    pinMode(GFX_BL, OUTPUT);
    digitalWrite(GFX_BL, LOW);
#endif

    gfx->setCursor(100, 100);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");

    delay(5000); // 5 seconds
}

void loop()
{
    gfx->setCursor(random(gfx->width()), random(gfx->height()));
    gfx->setTextColor(random(0xffff), random(0xffff));
    gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
    gfx->println("Hello World!");

    delay(1000); // 1 second`
getting an error 
"D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino:18:1: error: no matching function for call to 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int)'
   18 | );
      | ^
In file included from c:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:19,
                 from D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino:2:
c:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:77:3: note: candidate: 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, int32_t, bool, uint16_t, uint16_t)'
   77 |   Arduino_ESP32RGBPanel(
      |   ^~~~~~~~~~~~~~~~~~~~~
c:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:77:3: note:   candidate expects 33 arguments, 23 provided
c:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:74:7: note: candidate: 'constexpr Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(const Arduino_ESP32RGBPanel&)'
   74 | class Arduino_ESP32RGBPanel
      |       ^~~~~~~~~~~~~~~~~~~~~
c:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:74:7: note:   candidate expects 1 argument, 23 provided
c:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:74:7: note: candidate: 'constexpr Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(Arduino_ESP32RGBPanel&&)'
c:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:74:7: note:   candidate expects 1 argument, 23 provided
D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino:21:1: error: 'Arduino_ST7701_RGBPanel' does not name a type; did you mean 'Arduino_ESP32RGBPanel'?
   21 | Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
      | ^~~~~~~~~~~~~~~~~~~~~~~
      | Arduino_ESP32RGBPanel
D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino: In function 'void setup()':
D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino:30:5: error: 'gfx' was not declared in this scope
   30 |     gfx->begin();
      |     ^~~
D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino: In function 'void loop()':
D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino:48:5: error: 'gfx' was not declared in this scope
   48 |     gfx->setCursor(random(gfx->width()), random(gfx->height()));
      |     ^~~
D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino:53:17: error: expected '}' at end of input
   53 |     delay(1000); // 1 second
      |                 ^
D:\Yash\Relience Mixer griender\arduino\TFT_Display_try1\TFT_Display_try1\TFT_Display_try1.ino:47:1: note: to match this '{'
   47 | {
      | ^
Multiple libraries were found for "Adafruit_GFX.h"
  Used: C:\Users\YashChoudhary\AppData\Local\Arduino15\libraries\Adafruit-GFX-Library-master
  Not used: C:\Users\YashChoudhary\AppData\Local\Arduino15\libraries\Adafruit_GFX_Library
Using library SPI at version 3.0.7 in folder: C:\Users\YashChoudhary\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\SPI 
Using library GFX Library for Arduino at version 1.5.0 in folder: C:\Users\YashChoudhary\Documents\Arduino\libraries\GFX_Library_for_Arduino 
Using library Wire at version 3.0.7 in folder: C:\Users\YashChoudhary\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\Wire 
Using library Adafruit GFX Library at version 1.2.2 in folder: C:\Users\YashChoudhary\AppData\Local\Arduino15\libraries\Adafruit-GFX-Library-master 
exit status 1

Compilation error: no matching function for call to 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int)'
" 
i have taken references from below link
"https://forum.arduino.cc/t/esp32s3-and-st7701s/1004892"
also didn't find any library support in "Arduino_GFX_Library.h" for "ST7701S"

moononournation, 2024-12-18:
it is better follow example in PDQgraphicstest.ino instead of some code in forum

Yash333code, 2024-12-18:
didn't find any library support in "Arduino_GFX_Library.h" for "ST7701S"
Arduino_GFX_display.h does not have "ST7701S LCD" do i have to select any other LCD for the same, if yes then which one.

moononournation, 2024-12-18:
only have sample code for supported dev device

closed 2024-12-18 by Yash333code

FrankX0, 2025-05-06:
@Yash333code please give version 1.2.9 a try (if not resolved yet).
I was also using example code which made use of an old definition of Arduino_ESP32RGBPanel.