moononournation/Arduino_GFX #592: ESP32S3 with ST7701 - how to DMA in LVGL?

kchudoba, 2024-12-25:
I just wanted to ask if you know how to enable DMA on ESP32S3 with this library and LVGL? 
I managed to get LVGL working thanks to the examples but scrolling and transitions are very slow. 
Would appreciate any suggestions regarding performance improvement possibilities.

moononournation, 2024-12-26:
had you used `#define RGB_PANEL`?

closed 2024-12-26 by moononournation

kchudoba, 2024-12-26:
Here's my init code:

```
#define GFX_BL 38
Arduino_DataBus* bus = new Arduino_SWSPI(GFX_NOT_DEFINED /* DC */, 39 /* CS */, 48 /* SCK */, 47 /* MOSI */, GFX_NOT_DEFINED /* MISO */);

#define RGB_PANEL

Arduino_ESP32RGBPanel* rgbpanel = new Arduino_ESP32RGBPanel(
  18 /* DE */, 17 /* VSYNC */, 16 /* HSYNC */, 21 /* PCLK */,
  11 /* R0 */, 12 /* R1 */, 13 /* R2 */, 14 /* R3 */, 0 /* R4 */,
  8 /* G0 */, 20 /* G1 */, 3 /* G2 */, 46 /* G3 */, 9 /* G4 */, 10 /* G5 */,
  4 /* B0 */, 5 /* B1 */, 6 /* B2 */, 7 /* B3 */, 15 /* B4 */,
  1 /* hsync_polarity */, 10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
  1 /* vsync_polarity */, 10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);

Arduino_RGB_Display* gfx = new Arduino_RGB_Display(
  480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
  bus, GFX_NOT_DEFINED /* RST */, st7701_type9_init_operations, sizeof(st7701_type9_init_operations));
```

moononournation, 2024-12-26:
seems ok, you are using example of lvgl 8 or 9?

moononournation, 2024-12-26:
try v8