moononournation/Arduino_GFX #593: JC3248W535 AXS15231B LVGL Performance Problems

byte-me404, 2024-12-27:
Hello, I am using Arduino_GFX in combination with LVGL V9 on a [JC3248W535](https://www.aliexpress.com/item/1005007566332450.html) display with AXS15231B driver. 
I have the following problem: If I don't rotate the display I get about 26fps in the examples of LVGL when the image is still, when I scroll the frame rate drops to about 15fps which is still acceptable. If I rotate the display by 90° or 270°, the frame rate drops to 4fps when scrolling, which is no longer acceptable. I suspect that the reason for this lies in the implementation. I use a canvas to drive the display. As far as I understand it, the rotation is calculated in software, which makes the whole thing slow. Is there a simple solution to this problem? My code can be found [here](https://github.com/byte-me404/JC3248W535_lvgl_test/blob/main/src/main.cpp). 

Many thanks in advance :)

moononournation, 2024-12-28:
Yes, since AXS15231B not support rotation, so it requires software aids. It is not recommended rotation if need for speed.

byte-me404, 2024-12-28:
Okay, thanks for your quick replay. Do you know some LVGL/Arduino_GFX settings to speed up the display when used rotated?

closed 2025-01-05 by moononournation