moononournation/Arduino_GFX #594: Compile code error .isr_cpu_id

hvuitsme, 2024-12-28:
I comment out line .isr_cpu_id number 59 to temporarily ignore that error.

[image]

[image]

moononournation, 2024-12-29:
please use latest arduino-esp32

hvuitsme, 2024-12-29:
> please use latest arduino-esp32

Currently, there is no new update from your library or board manager, except for cloning from git.

moononournation, 2024-12-29:
arduino-esp32

h0lmes, 2024-12-29:
Searching for `arduino-esp32` in Arduino IDE Library Manager gives only `Arduino_ESP32_OTA` lib which is definitely not what you suggest.
What exactly do you mean by `arduino-esp32`?

Krishnawa, 2024-12-29:
I'm also facing the issue, so I downgraded the GFX_Library_for_Arduino to version 1.5.1.

moononournation, 2024-12-29:
ESP32 support should install in board manager

h0lmes, 2024-12-29:
`esp32` by Espressif Systems is recommended, so I'm using it.
Installing `Arduino ESP32 Boards` by Arduino didn't fix the issue.

moononournation, 2024-12-29:
update it

moononournation, 2024-12-29:
Latest release should be 3.1.0, released 2 weeks: https://github.com/espressif/arduino-esp32/releases/tag/3.1.0

closed 2024-12-29 by moononournation

h0lmes, 2024-12-29:
Ok, the latest version in Arduino IDE is still 3.0.7 but I see v3.1.0 is the latest on Github. Seems kinda bug from Arduino side.
@moononournation Thank you for pointing that out!

So I had to follow this guide https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html to add an Additional boards manager URL to the Arduino IDE.
After adding URL I closed and reopened Arduino IDE and it told me that I have updates to my boards. I let it do its thing and finally got v3.1.0 installed.
My project compiles successfully now.