moononournation/Arduino_GFX #595: Arduino_ESP32QSPI.cpp - how to compile?

real-skydiver, 2024-12-28:
I've updated my complete VSCode-environment to the latest versions. Still lots of compile-errors:
```
lib/Arduino_GFX/src/databus/Arduino_ESP32RGBPanel.cpp: In member function 'uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)':
lib/Arduino_GFX/src/databus/Arduino_ESP32RGBPanel.cpp:93:3: error: 'esp_lcd_rgb_panel_config_t::<unnamed struct>' has no non-static data member named 'refresh_on_demand'
   };
   ^
lib/Arduino_GFX/src/databus/Arduino_ESP32QSPI.cpp: In member function 'virtual bool Arduino_ESP32QSPI::begin(int32_t, int8_t)':
lib/Arduino_GFX/src/databus/Arduino_ESP32QSPI.cpp:59:21: error: 'ESP_INTR_CPU_AFFINITY_AUTO' was not declared in this scope
       .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\esp32s3box\libb62\Arduino_GFX\databus\Arduino_ESP32RGBPanel.cpp.o] Error 1
lib/Arduino_GFX/src/databus/Arduino_ESP32QSPI.cpp:59:21: note: suggested alternative: 'ESP_ETH_PHY_ADDR_AUTO'
       .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
                     ESP_ETH_PHY_ADDR_AUTO
lib/Arduino_GFX/src/databus/Arduino_ESP32QSPI.cpp:60:22: error: 'spi_bus_config_t' has no non-static data member named 'isr_cpu_id'
       .intr_flags = 0};
                      ^
lib/Arduino_GFX/src/databus/Arduino_ESP32QSPI.cpp:73:23: error: 'SPI_CLK_SRC_DEFAULT' was not declared in this scope
       .clock_source = SPI_CLK_SRC_DEFAULT,
                       ^~~~~~~~~~~~~~~~~~~
lib/Arduino_GFX/src/databus/Arduino_ESP32QSPI.cpp:73:23: note: suggested alternative: 'LCD_CLK_SRC_XTAL'
       .clock_source = SPI_CLK_SRC_DEFAULT,
                       ^~~~~~~~~~~~~~~~~~~
                       LCD_CLK_SRC_XTAL
lib/Arduino_GFX/src/databus/Arduino_ESP32QSPI.cpp:83:25: error: 'spi_device_interface_config_t' has no non-static data member named 'clock_source'
       .post_cb = nullptr};
                         ^
*** [.pio\build\esp32s3box\libb62\Arduino_GFX\databus\Arduino_ESP32QSPI.cpp.o] Error 1
```

The  ESP_INTR_CPU_AFFINITY_AUTO enum is defined in esp_intr_types.h which is part of the ESP-IDF > 5.3.x. - but how can I include this in my project?

moononournation, 2024-12-29:
Please use Arduino IDE

h0lmes, 2024-12-29:
@real-skydiver this reply might help  https://github.com/moononournation/Arduino_GFX/issues/594#issuecomment-2564715020

real-skydiver, 2024-12-29:
@h0lmes Thanks a million! I've already read all issues and the comments. However, I refuse to use "Arduino IDE" (I did so years ago - but working since way over 20yrs with VisualStudio (not "Code"!) - Arduino is no 'IDE' for me :)) 
I dumped Arduino_GFX altogether and use display drivers natively. At the end I'm using LVGL anyways most of the time. 
Anyways, once again, thanks for taking the time!!! 
 :)

closed 2025-01-05 by moononournation