moononournation/Arduino_GFX #599: Can you increase the driver for the HX8347G display screen?

liuquanli1970, 2025-01-05:
The initialization and rotation fragments of HX8347G are as follows:

#define TFT_GRAMWR         0x22 
#define TFT_COL_START_HIGH 0x02
#define TFT_COL_START_LOW  0x03
#define TFT_COL_END_HIGH   0x04
#define TFT_COL_END_LOW    0x05
#define TFT_ROW_START_HIGH 0x06
#define TFT_ROW_START_LOW  0x07
#define TFT_ROW_END_HIGH   0x08
#define TFT_ROW_END_LOW    0x09

void init();
{
  spi_end();
  delay(120);
  spi_begin();
  DC_C; tft_Write_8(0xEA); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0xEB); DC_D; tft_Write_8(0x20);
  DC_C; tft_Write_8(0xEC); DC_D; tft_Write_8(0x0C);
  DC_C; tft_Write_8(0xED); DC_D; tft_Write_8(0xC4);
  DC_C; tft_Write_8(0xE8); DC_D; tft_Write_8(0x40);
  DC_C; tft_Write_8(0xE9); DC_D; tft_Write_8(0x38);
  DC_C; tft_Write_8(0x27); DC_D; tft_Write_8(0xA3);
  DC_C; tft_Write_8(0x40); DC_D; tft_Write_8(0x01);

  DC_C; tft_Write_8(0x41); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0x42); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0x43); DC_D; tft_Write_8(0x10);
  DC_C; tft_Write_8(0x44); DC_D; tft_Write_8(0x0E);
  DC_C; tft_Write_8(0x45); DC_D; tft_Write_8(0x24);
  DC_C; tft_Write_8(0x46); DC_D; tft_Write_8(0x04);
  DC_C; tft_Write_8(0x47); DC_D; tft_Write_8(0x50);
  DC_C; tft_Write_8(0x48); DC_D; tft_Write_8(0x02);

  DC_C; tft_Write_8(0x49); DC_D; tft_Write_8(0x13);
  DC_C; tft_Write_8(0x4A); DC_D; tft_Write_8(0x19);
  DC_C; tft_Write_8(0x4B); DC_D; tft_Write_8(0x19);
  DC_C; tft_Write_8(0x4C); DC_D; tft_Write_8(0x16);
  DC_C; tft_Write_8(0x50); DC_D; tft_Write_8(0x1B);
  DC_C; tft_Write_8(0x51); DC_D; tft_Write_8(0x31);
  DC_C; tft_Write_8(0x52); DC_D; tft_Write_8(0x2F);
  DC_C; tft_Write_8(0x53); DC_D; tft_Write_8(0x3F);

  DC_C; tft_Write_8(0x54); DC_D; tft_Write_8(0x3F);
  DC_C; tft_Write_8(0x55); DC_D; tft_Write_8(0x3E);
  DC_C; tft_Write_8(0x56); DC_D; tft_Write_8(0x2F);
  DC_C; tft_Write_8(0x57); DC_D; tft_Write_8(0x7B);
  DC_C; tft_Write_8(0x58); DC_D; tft_Write_8(0x09);
  DC_C; tft_Write_8(0x59); DC_D; tft_Write_8(0x06);
  DC_C; tft_Write_8(0x5A); DC_D; tft_Write_8(0x06);
  DC_C; tft_Write_8(0x5B); DC_D; tft_Write_8(0x0C);

  DC_C; tft_Write_8(0x5C); DC_D; tft_Write_8(0x1D);
  DC_C; tft_Write_8(0x5D); DC_D; tft_Write_8(0xCC);
  DC_C; tft_Write_8(0x1B); DC_D; tft_Write_8(0x1B);
  DC_C; tft_Write_8(0x1A); DC_D; tft_Write_8(0x01);
  DC_C; tft_Write_8(0x24); DC_D; tft_Write_8(0x98);
  DC_C; tft_Write_8(0x25); DC_D; tft_Write_8(0x57);
  DC_C; tft_Write_8(0x23); DC_D; tft_Write_8(0x40);
  DC_C; tft_Write_8(0x18); DC_D; tft_Write_8(0x36);

  DC_C; tft_Write_8(0x19); DC_D; tft_Write_8(0x01);
  DC_C; tft_Write_8(0x1C); DC_D; tft_Write_8(0x06);
  DC_C; tft_Write_8(0x1F); DC_D; tft_Write_8(0x90);
  delay(10);
  DC_C; tft_Write_8(0x17); DC_D; tft_Write_8(0x05);
  DC_C; tft_Write_8(0x36); DC_D; tft_Write_8(0x08);
  DC_C; tft_Write_8(0x28); DC_D; tft_Write_8(0x38);
  delay(60);
  DC_C; tft_Write_8(0x28); DC_D; tft_Write_8(0x3C);
  DC_C; tft_Write_8(0x01); DC_D; tft_Write_8(0x00);

  DC_C; tft_Write_8(0x16); DC_D; tft_Write_8(0xA8);
  DC_C; tft_Write_8(0x02); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0x03); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0x04); DC_D; tft_Write_8(0x01);
  DC_C; tft_Write_8(0x05); DC_D; tft_Write_8(0x3F);
  DC_C; tft_Write_8(0x06); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0x07); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0x08); DC_D; tft_Write_8(0x00);
  DC_C; tft_Write_8(0x09); DC_D; tft_Write_8(0xEF);

  spi_end();
}

void setRotation(uint8_t rotation){

rotation = m % 4;      // Limit the range of values to 0-3
spi_begin();
switch(rotation){
  case 0:
      DC_C; tft_Write_8(0x16); DC_D; tft_Write_8(0x08);
      DC_C; tft_Write_8(0x04); DC_D; tft_Write_8(0x00);
      DC_C; tft_Write_8(0x05); DC_D; tft_Write_8(0xEF);
      DC_C; tft_Write_8(0x08); DC_D; tft_Write_8(0x01);
      DC_C; tft_Write_8(0x09); DC_D; tft_Write_8(0x3F);   
      _width  = _init_width;
      _height = _init_height;
      break;
  case 1:
      DC_C; tft_Write_8(0x16); DC_D; tft_Write_8(0xA8);
      DC_C; tft_Write_8(0x04); DC_D; tft_Write_8(0x01);
      DC_C; tft_Write_8(0x05); DC_D; tft_Write_8(0x3F);
      DC_C; tft_Write_8(0x08); DC_D; tft_Write_8(0x00);
      DC_C; tft_Write_8(0x09); DC_D; tft_Write_8(0xEF);   
      _width  = _init_height;
      _height = _init_width;
      break;
  case 2:
      DC_C; tft_Write_8(0x16); DC_D; tft_Write_8(0xC8);
      DC_C; tft_Write_8(0x04); DC_D; tft_Write_8(0x00);
      DC_C; tft_Write_8(0x05); DC_D; tft_Write_8(0xEF);
      DC_C; tft_Write_8(0x08); DC_D; tft_Write_8(0x01);
      DC_C; tft_Write_8(0x09); DC_D; tft_Write_8(0x3F);   
      _width  = _init_width;
      _height = _init_height;
      break;
  case 3:
      DC_C; tft_Write_8(0x16); DC_D; tft_Write_8(0x68);
      DC_C; tft_Write_8(0x04); DC_D; tft_Write_8(0x01);
      DC_C; tft_Write_8(0x05); DC_D; tft_Write_8(0x3F);
      DC_C; tft_Write_8(0x08); DC_D; tft_Write_8(0x00);
      DC_C; tft_Write_8(0x09); DC_D; tft_Write_8(0xEF);   
      _width  = _init_height;
      _height = _init_width;
      break;
}
spi_end();
}

moononournation, 2025-01-05:
I did not have this display in hand, but welcome you contribute a pull request.

liuquanli1970, 2025-01-05:
Thank you for your prompt reply. also thank you  for your effective Arduino GFX library. Your work is very meaningful and I greatly admire it.
HX8347G is an older LCD model that was widely used in some early devices(usually with SPI resistive touch screens).
I have some HX8347G LCD in hand, and I can help with testing and provide feedback on test results， If you can add this driver to the GFX library.
Thanks again.

liuquanli1970, 2025-01-05:
I use HX8347C driver, and the LCD can also work, but the color is reversed and (invertdisplay() does not work), and the left and right are mirrored displays.

liuquanli1970, 2025-01-05:
HX8347G is perfectly supported, I have made the following changes：

1.Added  ”#include“     in the  “Arduino_GFX_Library.h”
#include "display/Arduino_HX8347G.h"

2.Two files were created “Arduino_HX8347G.h”  and "Arduino_HX8347G.cpp" according to HX8347C  
Modified the following two functions:

void Arduino_HX8347G::tftInit()
{
  if (_rst != GFX_NOT_DEFINED)
  {
    pinMode(_rst, OUTPUT);
    digitalWrite(_rst, HIGH);
    delay(100);
    digitalWrite(_rst, LOW);
    delay(HX8347G_RST_DELAY);
    digitalWrite(_rst, HIGH);
    delay(HX8347G_RST_DELAY);
  }
  else
  {
    // Software Rest
  }

  // LCD Init For 3.0inch LCD Panel with HX8347G.
  // Power Voltage Setting
  _bus->sendCommand(0xEA); _bus->sendData(0x00);
  _bus->sendCommand(0xEB); _bus->sendData(0x20);
  _bus->sendCommand(0xEC); _bus->sendData(0x0C);
  _bus->sendCommand(0xED); _bus->sendData(0xC4);
  _bus->sendCommand(0xE8); _bus->sendData(0x40);
  _bus->sendCommand(0xE9); _bus->sendData(0x38);
  _bus->sendCommand(0x27); _bus->sendData(0xA3);
  _bus->sendCommand(0x40); _bus->sendData(0x01);

  _bus->sendCommand(0x41); _bus->sendData(0x00);
  _bus->sendCommand(0x42); _bus->sendData(0x00);
  _bus->sendCommand(0x43); _bus->sendData(0x10);
  _bus->sendCommand(0x44); _bus->sendData(0x0E);
  _bus->sendCommand(0x45); _bus->sendData(0x24);
  _bus->sendCommand(0x46); _bus->sendData(0x04);
  _bus->sendCommand(0x47); _bus->sendData(0x50);
  _bus->sendCommand(0x48); _bus->sendData(0x02);

  _bus->sendCommand(0x49); _bus->sendData(0x13);
  _bus->sendCommand(0x4A); _bus->sendData(0x19);
  _bus->sendCommand(0x4B); _bus->sendData(0x19);
  _bus->sendCommand(0x4C); _bus->sendData(0x16);
  _bus->sendCommand(0x50); _bus->sendData(0x1B);
  _bus->sendCommand(0x51); _bus->sendData(0x31);
  _bus->sendCommand(0x52); _bus->sendData(0x2F);
  _bus->sendCommand(0x53); _bus->sendData(0x3F);

  _bus->sendCommand(0x54); _bus->sendData(0x3F);
  _bus->sendCommand(0x55); _bus->sendData(0x3E);
  _bus->sendCommand(0x56); _bus->sendData(0x2F);
  _bus->sendCommand(0x57); _bus->sendData(0x7B);
  _bus->sendCommand(0x58); _bus->sendData(0x09);
  _bus->sendCommand(0x59); _bus->sendData(0x06);
  _bus->sendCommand(0x5A); _bus->sendData(0x06);
  _bus->sendCommand(0x5B); _bus->sendData(0x0C);

  _bus->sendCommand(0x5C); _bus->sendData(0x1D);
  _bus->sendCommand(0x5D); _bus->sendData(0xCC);
  _bus->sendCommand(0x1B); _bus->sendData(0x1B);
  _bus->sendCommand(0x1A); _bus->sendData(0x01);
  _bus->sendCommand(0x24); _bus->sendData(0x98);
  _bus->sendCommand(0x25); _bus->sendData(0x57);
  _bus->sendCommand(0x23); _bus->sendData(0x40);
  _bus->sendCommand(0x18); _bus->sendData(0x36);

  _bus->sendCommand(0x19); _bus->sendData(0x01);
  _bus->sendCommand(0x1C); _bus->sendData(0x06);
  _bus->sendCommand(0x1F); _bus->sendData(0x90);
  delay(10);
  _bus->sendCommand(0x17); _bus->sendData(0x05);
  _bus->sendCommand(0x36); _bus->sendData(0x08);
  _bus->sendCommand(0x28); _bus->sendData(0x38);
  delay(60);
  _bus->sendCommand(0x28); _bus->sendData(0x3C);
  _bus->sendCommand(0x01); _bus->sendData(0x00);

  _bus->sendCommand(0x16); _bus->sendData(0xA8);
  _bus->sendCommand(0x02); _bus->sendData(0x00);
  _bus->sendCommand(0x03); _bus->sendData(0x00);
  _bus->sendCommand(0x04); _bus->sendData(0x01);
  _bus->sendCommand(0x05); _bus->sendData(0x3F);
  _bus->sendCommand(0x06); _bus->sendData(0x00);
  _bus->sendCommand(0x07); _bus->sendData(0x00);
  _bus->sendCommand(0x08); _bus->sendData(0x00);
  _bus->sendCommand(0x09); _bus->sendData(0xEF);
}

void Arduino_HX8347G::setRotation(uint8_t r)
{
  Arduino_TFT::setRotation(r);
  switch (_rotation)
  {
  case 1:
      _bus->sendCommand(0x16); _bus->sendData(0xA8);
      _bus->sendCommand(0x04); _bus->sendData(0x01);
      _bus->sendCommand(0x05); _bus->sendData(0x3F);
      _bus->sendCommand(0x08); _bus->sendData(0x00);
      _bus->sendCommand(0x09); _bus->sendData(0xEF);   
    break;
  case 2:
      _bus->sendCommand(0x16); _bus->sendData(0xC8);
      _bus->sendCommand(0x04); _bus->sendData(0x00);
      _bus->sendCommand(0x05); _bus->sendData(0xEF);
      _bus->sendCommand(0x08); _bus->sendData(0x01);
      _bus->sendCommand(0x09); _bus->sendData(0x3F);   
    break;
  case 3:
      _bus->sendCommand(0x16); _bus->sendData(0x68);
      _bus->sendCommand(0x04); _bus->sendData(0x01);
      _bus->sendCommand(0x05); _bus->sendData(0x3F);
      _bus->sendCommand(0x08); _bus->sendData(0x00);
      _bus->sendCommand(0x09); _bus->sendData(0xEF);   
    break;
  default: // case 0:
      _bus->sendCommand(0x16); _bus->sendData(0x08);
      _bus->sendCommand(0x04); _bus->sendData(0x00);
      _bus->sendCommand(0x05); _bus->sendData(0xEF);
      _bus->sendCommand(0x08); _bus->sendData(0x01);
      _bus->sendCommand(0x09); _bus->sendData(0x3F);   
    break;
  }
}

3. Arduino_GFX *gfx =new Arduino_HX8347G();

Thank you again for your work！