moononournation/Arduino_GFX #601: Elecrow TFT 4.3 inch

yaya54840, 2025-01-07:
Hello
I would like to ask you if you know this screen
with display NV3047  https://www.amazon.fr/dp/B0CD1HW5XT?ref=emc_s_m_5_i_atc
I managed to make it work on lovyan_gfx by including wire and spi.
but does not work on your library nor on tft_espi.
If you need more information.
Waiting for an answer

moononournation, 2025-01-07:
I cannot suport a dev device not in hand, but it is welcome for your contribution on a pull request about it.

yaya54840, 2025-01-07:
Thank you for your quick response

yaya54840, 2025-01-07:
sorry Thank you for your quick response

Shadowtrance, 2025-01-07:
Looks like this is the same display. There's an LGFX config there and an Arduino GFX config there too.
Worth a try! :)
https://www.elecrow.com/wiki/esp32-display-432727-intelligent-touch-screen-wi-fi26ble-480272-hmi-display.html

closed 2025-01-07 by moononournation