moononournation/Arduino_GFX #603: Compilation Error – Arduino_ESP32LCD8 not recognized

w569111843, 2025-01-09:
```
Description: I'm trying to use the Arduino_ESP32LCD8 class from the Arduino_GFX_Library, but I am encountering a compilation error stating that Arduino_ESP32LCD8 is not `recognized.`

Error Message:
go
复制代码
D:\arduino-project\sketch_jan9a\PINS_ESP32-S3-CAM.h:4:28: error: expected type-specifier before 'Arduino_ESP32LCD8'
    4 | Arduino_DataBus *bus = new Arduino_ESP32LCD8(
      |                            ^~~~~~~~~~~~~~~~~
Steps to Reproduce:
Include the Arduino_GFX_Library in the sketch.
Use the following code to create an Arduino_ESP32LCD8 object:
cpp
复制代码
#include <Arduino_GFX_Library.h>

Arduino_DataBus *bus = new Arduino_ESP32LCD8(
    TFT_DC, 27, 57 /* WR */, 45 /* RD */,
    51 /* D0 */, 50 /* D1 */, 48 /* D2 */, 47 /* D3 */, -1 /* D4 */, -1 /* D5 */, -1 /* D6 */, -1 /* D7 */);
Arduino_GFX *gfx = new Arduino_ST77916(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);
Compile the sketch.
Expected Behavior:
The sketch should compile without errors, recognizing the Arduino_ESP32LCD8 class.

Actual Behavior:
The compiler throws an error:

go
复制代码
expected type-specifier before 'Arduino_ESP32LCD8'
Additional Information:
I have included Arduino_ESP32LCD8.h within the Arduino_GFX_Library.h, and it is properly referenced.
The library Arduino_GFX_Library is installed through the Arduino Library Manager.
I am using ESP32-S3 as the target platform.
I have verified the code on other ESP32 boards, but the error persists specifically on ESP32-S3.
Possible Causes:
The Arduino_ESP32LCD8 class might not be recognized due to an issue with the library installation or the inclusion path.
There might be a compatibility issue with the ESP32-S3 platform or the version of the library.
System Details:
Arduino IDE Version: [e.g., 2.3.4]
ESP32 Version: [e.g., 3.1.1]
Operating System: [e.g., Windows 11]
ESP32 Board: ESP32-S3
Suggestions:
Please check if the class Arduino_ESP32LCD8 is supported on ESP32-S3, or if the library version has any compatibility issues.
Ensure that the required dependencies and includes are correctly handled.

```

Big-mouth-c, 2025-01-10:
Note
ESP32LCD8, ESP32LCD16 and ESP32RGBPanel only supported by arduino-esp32 v2.x and no longer support in v3.0.

Big-mouth-c, 2025-01-10:
arduino-esp32 v3.x
ESP32LCD8->ESP32PAR8

w569111843, 2025-01-10:
OK I'll try

w569111843, 2025-01-10:
```
Arduino_DataBus *bus = new Arduino_ESP32PAR8(40 /* DC */, 21 /* CS */, -1 /* WR */, GFX_NOT_DEFINED /* RD */,
                                             46 /* d0 */, 45 /* d1 */, 42 /* d2 */, 41 /* d3 */, -1 /* D4 */, -1 /* D5 */, -1 /* D6 */, -1 /* D7 */);
Arduino_ST77916 *gfx = new Arduino_ST77916(bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */, true /* IPS */, 360 /* width */, 360 /* height */);
```
I used Arduino_ESP32QSPI, but the video cannot display images. I'm not sure if there is an issue with the pins I used. I'm a beginner and don't understand much. Could you please help me check? Here is the schematic [ESP32-S3-LCD-1.85.pdf](https://www.waveshare.net/w/upload/7/76/ESP32-S3-LCD-1.85.pdf)

w569111843, 2025-01-10:
```
Arduino_DataBus *bus = new Arduino_ESP32QSPI(
    21 /* cs */, 40 /* sck */, 46 /* d0 */, 45 /* d1 */, 42 /* d2 */, 41 /* d3 */);
Arduino_ST77916 *gfx = new Arduino_ST77916(bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */, true /* IPS */, 360 /* width */, 360 /* height */);
```
Here is the code I used with Arduino_ESP32QSPI.

Big-mouth-c, 2025-01-11:
#define GFX_BL
Arduino_DataBus *bus = new Arduino_ESP32QSPI( 21 /* CS */, 40 /* SCK */, 46 /* D0 */, 45 /* D1 */, 42 /* D2 */, 41 /* D3 */);
Arduino_GFX *gfx = new Arduino_ST77916(bus, -1/* RST */, 0 /* rotation */, false /* IPS */);

Big-mouth-c, 2025-01-11:
LCD引脚 | ESP32S3

LCD_SDA0 | GPIO46
LCD_SDA1 | GPIO45
LCD_SDA2 | GPIO42
LCD_SDA3 | GPIO41
LCD_SCK | GPIO40
LCD_CS | GPIO21
LCD_TE | GPIO18
LCD_RST | EXIO2
LCD_BL | GPIO5
TP_SDA | GPIO1
TP_SCL | GPIO3
TP_INT | GPIO4
TP_RST | EXIO1

w569111843, 2025-01-11:
Yes, with this configuration, I can light up the screen and play the video. However, after the video finishes playing and I call avi_close(), it crashes

w569111843, 2025-01-11:
I found that when executing the AVI_close method, freeing the memory with free(AVI->idx); causes an error.

moononournation, 2025-01-11:
Please note the avilib and avi player sample is not provided in this repository. This should not raise an issue in Arduino_GFX. I suggest you may try another avi file, most probably the avi provided in sample code, to check problem still presist.

closed 2025-01-11 by w569111843