moononournation/Arduino_GFX #606: Compiler error

RolandRust, 2025-01-11:
Hi, thx for the great work.

I get now two compiler errors:

RolandRust, 2025-01-11:
error: 'ESP_INTR_CPU_AFFINITY_AUTO' was not declared in this scope
and 
error: 'spi_device_interface_config_t' has no non-static data member named 'clock_source'

if I comment in the file: GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp
the lines: 
59: .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
and
73: .clock_source = SPI_CLK_SRC_DEFAULT,

then it works, and in a short test it looks like the Display is making what he should.

I am using:
Chip: ESP32 and ESP32-S3
IDE: Visual Studio Code 1.96.2
PlatformIO: 6.1.16
Platform: Espressif 32 6.9.0

ThomDyson, 2025-01-12:
Version 1.5.x of GFX requires the board definition ESP32 of version 3.1.0 or greater. You'll need to update that file set.   I (reluctantly) moved back to Arduino IDE, where getting the latest board definitions is pretty easy.
Depending on your display, you may or may not need that section of code to work.

moononournation, 2025-01-14:
Please do not raise any PIO related error here, you should report it to PIO responsible parties.

closed 2025-01-14 by moononournation