moononournation/Arduino_GFX #609: ESP32S3 Random horizontal shift of the image after init on 1.5.2

Suprazz, 2025-01-14:
Dev board is MaTouch ESP32-S3 Rotary IPS Display with Touch 2.1“ ST7701

The demo provided by Makerfab works on version Arduino_GFX 1.3.1.
I wanted to update to the latest esp32 library so I was able to manage and get everything working but I have one issue left.

I'm not sure when it happen or why but with GFX 1.5.2, ESP 3.1.1 and LVGL 8.3 after Display init, most of the time, the image is shifted horizontally. Usually if I reset the board few times, eventually it will be ok and the image will be aligned with the display.

The display works perfectly and there are no artefacts. Only the image is not centered with the display and overlap on the other side. I can execute my application, switch to different Lvgl screens and touch events are ok, only the image is offset horizontally from the actual display and never vertically.

I checked all the init sequence and everything is the same as before.

Any suggestions on what should I look for? I searched but can't figure out why

I think it's something corrupted that is sent while doing the init sequence. Tried to read the ST7701S datasheet but couldn't figure it out.

`Arduino_DataBus* bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, 1 /* CS */,
    46 /* SCK */, 0 /* MOSI */, GFX_NOT_DEFINED /* MISO */);

Arduino_ESP32RGBPanel* rgbpanel = new Arduino_ESP32RGBPanel(
    2 /* DE */, 42/* VSYNC */, 3 /* HSYNC */, 45 /* PCLK */,
    4 /* R0 */, 41 /* R1 */, 5 /* R2 */, 40 /* R3 */, 6 /* R4 */,
    39 /* G0/P22 */, 7 /* G1/P23 */, 47 /* G2/P24 */, 8 /* G3/P25 */, 48 /* G4/P26 */, 9 /* G5 */,
    11 /* B0 */, 15 /* B1 */, 12 /* B2 */, 16 /* B3 */, 21 /* B4 */,
    1 /* hsync_polarity */, 10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
    1 /* vsync_polarity */, 10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);

Arduino_RGB_Display* gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, st7701_type5_init_operations, sizeof(st7701_type5_init_operations));`

ThomDyson, 2025-01-14:
I think you need to post more of you code showing how LVGL and the GFX library interact. I did a quick search and found suggestions that this is a LVGL issue. What's your display buffer size?

Suprazz, 2025-01-14:
Sure. Here is the config of the display with lvgl. I must say that while the program it's running it never goes back to the centered position unless we do a reset and it might be ok or not. It's really random. I tried to add delays before the init but it didnt change anything.

`
gfx->begin();
gfx->fillScreen(RGB565_BLACK);
lv_init();   
#define DRAW_BUF_SIZE (EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES) // 480x480

 lv_color_t *buf1 = (lv_color_t *)heap_caps_malloc(DRAW_BUF_SIZE * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
    assert(buf1);
    lv_color_t *buf2 = (lv_color_t *)heap_caps_malloc(DRAW_BUF_SIZE * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
    assert(buf2);
    lv_disp_draw_buf_init(&disp_buf, buf1, buf2, DRAW_BUF_SIZE * sizeof(lv_color_t));

    DEBUG_SERIAL_BOOT(TAG, "Register display driver to LVGL");

    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = EXAMPLE_LCD_H_RES;
    disp_drv.ver_res = EXAMPLE_LCD_V_RES;
    disp_drv.flush_cb = lvgl_flush_cb;
    disp_drv.draw_buf = &disp_buf;
    disp_drv.user_data = panel_handle;
    lv_disp_t *disp = lv_disp_drv_register(&disp_drv);
    if (disp == NULL)
    {
        DEBUG_SERIAL_BOOT(TAG, "DISP INIT ERROR");
        while (1);
    }`

And the flush fonction:
`    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

    vTaskSuspendAll();
#if (LV_COLOR_16_SWAP != 0)
    gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t*)&color_map->full, w, h);
#else
    gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t*)&color_map->full, w, h);
#endif
    xTaskResumeAll(); 
lv_disp_flush_ready(drv);`

Suprazz, 2025-01-14:
After more testing I noticed a random vertical shift and found other problems. That might be related... it's mostly caused by esp32 3.1.1

ThomDyson, 2025-01-14:
When you say mostly caused by 3.1.1, you mean the complete set of your current versions? Or was there a different set of GFX 1.5.x, LVGL8.3 and other esp32?  Are you sure EXAMPLE_LCD_H_RES is set right? Have tried not using double buffering?
Take a look at this https://www.esp32.com/viewtopic.php?t=32996

Suprazz, 2025-01-14:
Thanks for the link I'll explore this. It could explain few things. I'll report back with what I find

ThomDyson, 2025-01-14:
BTW, I am using the exact same hardware, with GFX 1.5.1 but LVGL 9.  Everything centers fine.

ThomDyson, 2025-01-16:
Fixed my flicker problem.  I had trimmed a lot of code and messed up the  my_disp_flush function.  I've edited it so it works.

ThomDyson, 2025-01-16:
I draw in the main loop because the sample code does. Drawing in the flush function caused my flickering.

closed 2025-01-22 by moononournation