moononournation/Arduino_GFX #611: GFX Library with platformio and esp32-s3-devkitm-1

KimSchneider, 2025-01-18:
Hi, 

I really like your library. I got sample code with my ESP32-1732S019 and that works in Arduino IDE. However I am used to platform.io more. 
For some reason the same code does not work in platform.io but works in Arduino IDE. I seem to be missing some dependency in platform.io that Arduino IDE has in the background. 

Could you please provide a hint what the issue could be? 

Config results in Arduino@2.0.17
```[env:esp32-s3-devkitm-1]
platform = espressif32@6.7.0
board = esp32-s3-devkitm-1
framework = arduino
monitor_speed = 115200
lib_deps = moononournation/GFX Library for Arduino@^1.5.2
```

Error is
```Compiling .pio\build\esp32-s3-devkitm-1\lib6b9\GFX Library for Arduino\databus\Arduino_ESP32S2PAR8Q.cpp.o
Compiling .pio\build\esp32-s3-devkitm-1\lib6b9\GFX Library for Arduino\databus\Arduino_ESP32SPI.cpp.o
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp: In member function 'virtual bool Arduino_ESP32QSPI::begin(int32_t, int8_t)':
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:59:21: error: 'ESP_INTR_CPU_AFFINITY_AUTO' was not declared in this scope
       .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:59:21: note: suggested alternative: 'ESP_ETH_PHY_ADDR_AUTO'
       .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
                     ESP_ETH_PHY_ADDR_AUTO
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:60:22: error: 'spi_bus_config_t' has no non-static data member named 'isr_cpu_id' 
       .intr_flags = 0};
                      ^
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:73:23: error: 'SPI_CLK_SRC_DEFAULT' was not declared in this scope
       .clock_source = SPI_CLK_SRC_DEFAULT,
                       ^~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:73:23: note: suggested alternative: 'LCD_CLK_SRC_XTAL'
       .clock_source = SPI_CLK_SRC_DEFAULT,
                       ^~~~~~~~~~~~~~~~~~~
                       LCD_CLK_SRC_XTAL
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:83:25: error: 'spi_device_interface_config_t' has no non-static data member named 'clock_source'
       .post_cb = nullptr};
                         ^
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.cpp: In member function 'uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)':
.pio/libdeps/esp32-s3-devkitm-1/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.cpp:93:3: error: 'esp_lcd_rgb_panel_config_t::<unnamed struct>' has no non-static data member named 'refresh_on_demand'
   };
   ^
*** [.pio\build\esp32-s3-devkitm-1\lib6b9\GFX Library for Arduino\databus\Arduino_ESP32QSPI.cpp.o] Error 1
*** [.pio\build\esp32-s3-devkitm-1\lib6b9\GFX Library for Arduino\databus\Arduino_ESP32RGBPanel.cpp.o] Error 1
```

moononournation, 2025-01-18:
Please report pio related issue to responsible parties.

closed 2025-01-22 by moononournation

KimSchneider, 2025-01-22:
I found help in the platform io forum. If anyone has a smiliar issue, [here is the solution](https://community.platformio.org/t/gfx-library-with-platformio-on-esp32-1732s019/45461).