moononournation/Arduino_GFX #613: ESP32 and WEA2012 not working in PDQgraphicstest.ino example

M-Allahham, 2025-01-21:
When I uncomment the last 3 lines in Arduino_GFX_display.h and just use the default databus and pins, I get the following output:

Arduino_GFX PDQgraphicstest example!
gfx->begin() failed!
Benchmark	micro-secs

followed by a core panic. 

Is there a guide or something I can follow to get the WEA2012 display working?

ThomDyson, 2025-01-22:
Note that for the test, you either uncomment lines inArduino_GFX_dev_device.h or rely on the other files. Not both.  
Make sure you have the right Board type set in the IDE - ESP32 Dev Module.
The code you've provided doesn't match the sample in the Arduino_GFX_display.h  That display uses the CANVAS method to interact with the display.  Note that the last 3 lines of GFX_display still expect you to define the "Arduino_DataBus *bus". This is in the Arduino_GFX_Databus.h  That file keys off the Board type you've selected.

moononournation, 2025-01-22:
If reboot while initial the QSPI, in most case it is used the GPIO that cannot use. E.g. the GPIO connected to PSRAM.

closed 2025-01-22 by moononournation