moononournation/Arduino_GFX #614: Ignored x and y coordinates

adam-bubela, 2025-01-22:
No matter what I put into `fillRect()` the rectangle always starts at 0px, 0px (the top-right corner).

for example:
`gfx->fillRect(160, 240, 160, 240, MAGENTA);`

results 
[image]

my code
```
#include <Arduino.h>
#include <Arduino_GFX_Library.h>
#define GFX_BL 1

void setup(){
  Arduino_DataBus *bus = new Arduino_ESP32QSPI(
    45 /* CS */, 
    47 /* SCK */, 
    21 /* D0 */, 
    48 /* D1 */, 
    40 /* D2 */, 
    39 /* D3 */
  );
    
  Arduino_GFX *gfx = new Arduino_AXS15231B(
    bus, 
    GFX_NOT_DEFINED /* RST */, 
    0 /* rotation */, 
    false /* IPS */, 
    320 /* width */,
    480 /* height*/
  );

  if(!gfx->begin())
    Serial.println("gfx->begin() failed!");
  
  gfx->fillScreen(RGB565_GREEN);
  #ifdef GFX_BL
      pinMode(GFX_BL, OUTPUT);
      digitalWrite(GFX_BL, HIGH);
  #endif
  gfx->fillRect(160, 240, 160, 240, random(0,64000)); //rectangle pulled to the top-right corner
  gfx->drawRect(200, 300, 100, 90, ORANGE); //shows nothing
}
```
Feels like I may be making some silly error though :-)

ThomDyson, 2025-01-22:
That looks like your rotation is wrong. What does  

`gfx->fillRect(0, 0, 160, 240, random(0,64000)); `

look like?

adam-bubela, 2025-01-22:
> That looks like your rotation is wrong. What does
> 
> `gfx->fillRect(0, 0, 160, 240, random(0,64000)); `
> 
> look like?

exactly the same position

[image]

adam-bubela, 2025-01-22:
Now, if I only add `Arduino_Canvas` support  the display works as expected:
```
Arduino_GFX *g = new Arduino_AXS15231B(bus, GFX_NOT_DEFINED, 0 , false , 320, 480);
Arduino_Canvas *gfx  = new Arduino_Canvas(320, 480, g, 0, 0, 0);

gfx->fillScreen(RGB565_GREEN);

for (int y = 0; y < 480; y+=10){
      gfx->fillRect(0, y, 160, 240, random(0,64000));
      gfx->fillCircle(160, y, 80, random(0,64000));
      gfx->flush();
      Serial.printf("pos y:%d px\n",y);
      delay(1000);
}
```

[image]

moononournation, 2025-01-22:
If you found a way to make rotation for AXS15231B, welcome to make a pull request.

adam-bubela, 2025-01-23:
> If you found a way to make rotation for AXS15231B, welcome to make a pull request.

The current library rotates my AXS15231B display as expected if I use `Arduino_Canvas`.