moononournation/Arduino_GFX #616: Screen keeps scrolling up on use when using SPIRAM

milGus, 2025-01-24:
demo widgets using lvgl 9.2.2 and the latest Arduino_GFX and ST7262, the screen keeps scrolling up when touched when using

  disp_draw_buf = (lv_color_t *) heap_caps_malloc(buf_size_in_bytes, MALLOC_CAP_SPIRAM);
  disp_draw_buf2 = (lv_color_t *) heap_caps_malloc(buf_size_in_bytes, MALLOC_CAP_SPIRAM);

It doesnt scroll when 

  	disp_draw_buf = (lv_color_t *) heap_caps_malloc(buf_size_in_bytes, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);
	disp_draw_buf2 = (lv_color_t *) heap_caps_malloc(buf_size_in_bytes, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);

or even

    disp_draw_buf = (lv_color_t *) heap_caps_malloc(buf_size_in_bytes, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
    disp_draw_buf2 = (lv_color_t *) heap_caps_malloc(buf_size_in_bytes, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);

Any ideas?

milGus, 2025-01-29:
any help?

moononournation, 2025-01-30:
Not tried LVGL double buffer a while, because this setting only make it slower. You may try single buffer first.

milGus, 2025-01-30:
Yeah I tried that...no go. However Ive noticed even with internal ram it does it. but it takes a LONG time like 20mins before it happens...but a lot shorter scroll...like a few pixels at a time...so by 1 hour or so its shifted 45pixels.