moononournation/Arduino_GFX #618: Changes to Espressif framework produce compile error

Starbane, 2025-01-26:
The library uses deprecated material

GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:59:21: error: 'ESP_INTR_CPU_AFFINITY_AUTO' was not declared in this scope
       .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~

GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.cpp:93:3: error: 'esp_lcd_rgb_panel_config_t::<unnamed struct>' has no non-static data member named 'refresh_on_demand'

GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:60:22: error: 'spi_bus_config_t' has no non-static data member named 'isr_cpu_id'
       .intr_flags = 0};

GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:73:23: error: 'SPI_CLK_SRC_DEFAULT' was not declared in this scope
       .clock_source = SPI_CLK_SRC_DEFAULT,
GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:73:23: note: suggested alternative: 'LCD_CLK_SRC_XTAL'
       .clock_source = SPI_CLK_SRC_DEFAULT,
                       ^~~~~~~~~~~~~~~~~~~
                       LCD_CLK_SRC_XTAL
GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:83:25: error: 'spi_device_interface_config_t' has no non-static data member named 'clock_source'
       .post_cb = nullptr};

This breaks every version of Arduino_GFX_Library that uses PlatformIO with anyone of the latest Espressif toolchain and framework.

In dire need of a working library!

Starbane, 2025-01-26:
Forgot to mention platforms tried! I suspect though that perhaps not all links to the platform is transfered when you switch so take this list with a pinch of salt! I would think that 3.0.0 would actually work.

Platforms
├── espressif32 @ 6.9.0 (required: platformio/espressif32)
├── espressif32 @ 3.0.0 (required: platformio/espressif32)
├── espressif32 @ 3.3.0 (required: platformio/espressif32)
├── espressif32 @ 3.4.0 (required: platformio/espressif32)
├── espressif32 @ 4.4.0 (required: platformio/espressif32)
├── espressif32 @ 5.4.0 (required: platformio/espressif32)
├── espressif32 @ 6.5.0 (required: platformio/espressif32)
├── espressif32 @ 6.6.0 (required: platformio/espressif32)
└── espressif32 @ 6.9.0+sha.e972aab (required: git+https://github.com/platformio/platform-espressif32.git)

ThomDyson, 2025-01-26:
I think you have this backwards. ESP_INTR_CPU_AFFINITY_AUTO is not deprecated. It is part of the current ESP IDF (5.3.x). Platformio doesn't natively use the current version. The Arduino IDE is the supported IDE. You'll have to look for help to see how/if platformio can be forced to use the current code versions.

Starbane, 2025-01-26:
Thanks for taking the time to answer!

I tried a quick test and switched to a version that you suggested (5.3.1) with the command in platformio.ini

platform_packages =
  ***@***.***://github.com/espressif/esp-idf.git <https://github.com/espressif/esp-idf.git#v5.3.1>#v5.3.1 <https://github.com/espressif/esp-idf.git#v5.3.1>

But it ended up with almost the same errors. So I am thinking there is something other that is causing this. The frustrating thing is that I have had this working for years, invested in using your library and when I (after like 6 months) of doing other stuff I get back and can’t get any of my code working. My thinking is that if I can’t get this to work, there got to be more out there that are having the same trouble. For me Arduino is not and IDE but rather a IDF. Spreading you library using the same IDF to other IDE’s I would think would promote your library, perhaps I am wrong?

moononournation, 2025-01-27:
Arduino_GFX only support Arduino IDE, you should report pio related issue to responsible parties

ThomDyson, 2025-01-29:
Vathivis - update your esp32 version under Board Manager.

moononournation, 2025-01-30:
Try commit a fix to master branch, please get update and test again.

Starbane, 2025-01-30:
I can recommend another platform for PlatformIO. Checkout pioarduino. Chance you platformio.ini file to use another platform like so:

platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip

Did compile but banged on graphics (all black screen)

closed 2025-02-06 by moononournation