moononournation/Arduino_GFX #620: NV3052C-GRB 720x720 Display

AAJAY5, 2025-01-31:
# **Display not working or not initializing properly**. As the same code with different display works like st7701s.

## Display 
- I have [TT395TFC04B](https://www.alibaba.com/product-detail/3-95-inch-720-720-SPI_1601264210385.html)  display. So, as per the provided information display have `NV3052` driver IC
- Display [Link1](https://www.toppoplcd.com/productdetails_5701775.html) &  [Link2](https://www.alibaba.com/product-detail/3-95-inch-720-720-SPI_1601264210385.html)

## Board
- I have [ESP32-S3 RGB LCD Driver Board](https://thinkrobotics.com/products/esp32-s3-rgb?variant=49312240402749&country=IN&currency=INR&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&srsltid=AfmBOopCHNbZkieG8jJ_2SrWKZGsM7h8dLIVhuLeNn5nkREY0zcX9JHr1b8&com_cvv=d30042528f072ba8a22b19c81250437cd47a2f30330f0ed03551c4efdaf3409e)
- Display and Board are pin to pin compatible

## Code

- Used `hd40015c40_init_operations` init sequence. As [this](https://www.adafruit.com/product/5793) display have same init seq and same driver. Difference is just shape.

```
#include <lvgl.h>
#include <demos/lv_demos.h>
#include <Arduino_GFX_Library.h>

Arduino_DataBus *bus = new Arduino_SWSPI(
  GFX_NOT_DEFINED /* DC */,
  42 /* CS */,
  2 /* SCK */,
  1 /* MOSI */,
  GFX_NOT_DEFINED /* MISO */
);

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  40 /* DE */,
  39 /* VSYNC */,
  38 /* HSYNC */,
  41 /* PCLK */,
  46 /* R0 */, 3 /* R1 */, 8 /* R2 */, 18 /* R3 */, 17 /* R4 */,
  14 /* G0 */, 13 /* G1 */, 12 /* G2 */, 11 /* G3 */, 10 /* G4 */, 9 /* G5 */,
  5 /* B0 */, 45 /* B1 */, 48 /* B2 */, 47 /* B3 */, 21 /* B4 */,
  1 /* hsync_polarity */,
  10 /* hsync_front_porch */,
  10 /* hsync_pulse_width */,
  10 /* hsync_back_porch */,
  1 /* vsync_polarity */,
  14 /* vsync_front_porch */,
  2 /* vsync_pulse_width */,
  12 /* vsync_back_porch */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
  720 /* width */,
  720 /* height */,
  rgbpanel,
  0 /* rotation */,
  true /* auto_flush */,
  bus,
  GFX_NOT_DEFINED /* RST */,
  hd40015c40_init_ooperations, sizeof(hd40015c40_init_operations));

// #endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

/*******************************************************************************
 * Please config the touch panel in touch.h
 ******************************************************************************/
#include "touch.h"

uint32_t screenWidth;
uint32_t screenHeight;
uint32_t bufSize;
lv_disp_draw_buf_t draw_buf;
lv_color_t *disp_draw_buf;
lv_disp_drv_t disp_drv;

#if LV_USE_LOG != 0
/* Serial debugging */
void my_print(const char *buf) {
  Serial.printf(buf);
  Serial.flush();
}
#endif

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p) {

  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);

#if (LV_COLOR_16_SWAP != 0)
  gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#else
  gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#endif

  gfx->flush();
  lv_disp_flush_ready(disp_drv);
}

static void example_increase_lvgl_tick(void *arg) {
  /* Tell LVGL how many milliseconds has elapsed */
  lv_tick_inc(2);
}

/*Read the touchpad*/
void my_touchpad_read(lv_indev_drv_t *indev_drv, lv_indev_data_t *data) {
  if (touch_has_signal()) {
    if (touch_touched()) {
      data->state = LV_INDEV_STATE_PR;

      /*Set the coordinates*/
      data->point.x = touch_last_x;
      data->point.y = touch_last_y;
    } else if (touch_released()) {
      data->state = LV_INDEV_STATE_REL;
    }
  } else {
    data->state = LV_INDEV_STATE_REL;
  }
}

void setup() {
  Serial.begin(115200);
  // Serial.setDebugOutput(true);
  // while(!Serial);
  Serial.print("Arduino_GFX LVGL Benchmark example ");
  String LVGL_Arduino = String('V') + lv_version_major() + "." + lv_version_minor() + "." + lv_version_patch();
  Serial.println(LVGL_Arduino);

  // Init Display
  if (!gfx->begin()) {
    Serial.println("gfx->begin() failed!");
  }
  gfx->fillScreen(RGB565_BLACK);

#ifdef GFX_BL
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
#endif

  // Init touch device
  touch_init(gfx->width(), gfx->height(), gfx->getRotation());

  lv_init();

  screenWidth = gfx->width();
  screenHeight = gfx->height();

  ESP_LOGI(TAG, "Initialize LVGL library");
  void *buf1 = NULL;
  void *buf2 = NULL;

  ESP_LOGI(TAG, "Allocate separate LVGL draw buffers from PSRAM");
  buf1 = heap_caps_malloc(screenWidth * screenHeight * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
  assert(buf1);
  buf2 = heap_caps_malloc(screenWidth * screenHeight * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
  assert(buf2);
  // initialize LVGL draw buffers
  lv_disp_draw_buf_init(&draw_buf, buf1, buf2, screenWidth * screenHeight);

  ESP_LOGI(TAG, "Register display driver to LVGL");
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = screenWidth;
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  // disp_drv.user_data = panel_handle;
  // #if CONFIG_EXAMPLE_DOUBLE_FB
  disp_drv.full_refresh = true;  // the full_refresh mode can maintain the synchronization between the two frame buffers
                                 // #endif
  lv_disp_t *disp = lv_disp_drv_register(&disp_drv);

  /* Initialize the (dummy) input device driver */
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read;
  lv_indev_drv_register(&indev_drv);

  lv_demo_widgets();
  // lv_demo_keypad_encoder();
  // lv_demo_benchmark();
  // lv_demo_stress();
  // lv_demo_music();

  const esp_timer_create_args_t lvgl_tick_timer_args = {
    .callback = &example_increase_lvgl_tick,
    .name = "lvgl_tick"
  };
  esp_timer_handle_t lvgl_tick_timer = NULL;
  ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
  ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, 2 * 1000));
  Serial.println("Setup done");
}

void loop() {
  lv_timer_handler(); /* let the GUI do its work */
  delay(10);
}

```

moononournation, 2025-01-31:
Sorry, I cannot support the display not in hand. You may try to apply the settings according to your display data sheet.

ThomDyson, 2025-01-31:
You have a typo with this hd40015c40_init_ooperations.  Too many "o".

closed 2025-02-06 by moononournation