moononournation/Arduino_GFX #625: Colour Defaults in ST7701 (RGBPanel) are inverted and no InvertDisplay function included after version 1.3.1

dagnall53, 2025-02-14:
In your update from 1.3.1 you refactored (and renamed!) the panel displays drivers, 
But in updating the ST7701 defines (for some reason now called  RGBpanel?) you have removed the ips from the setup and also removed the invertDisplay() and rotate() functions.
As noted before, the default is now colour inverted and without any invertDisplay() function we cannot get correct colours displayed.

The quick answer you gave some time ago to another user was to use the modified version of your library that the manufacturers distribute. but this prevents us from using your more recent versions. 

The InvertDisplay() function worked perfectly fine in 1.3.1 and so, Can you please add it back ?
I have tried to add the older version, but the code doe not know about _ips ! 
It must be as simple as writing to the ST7701, as this is what you did in 1.3.1 - but unfortunately I do not know what addresses you wrote to! 
I hoping you can add the (working!) equivalent to ..
void Arduino_RGB_Display::invertDisplay(bool i)
{
  _bus->sendCommand(_ips ? (i ? 0x20 : 0x21) : (i ? 0x21 : 0x20));
} 

I have tried this, but _ips is not defined, so it does not compile, but I presume you could hard code the correct address for the _ips? 

Many thanks
Dagnall

ThomDyson, 2025-02-15:
I think the idea is that something like 'inverted' colors is a configuration setting, not something that should be a function.  If you look at the settings in src/display/Arduino_RGB_Display.h there are several variations on 7701 configurations.  The comments on the 7701 types do mention ips and the two hex values.    The example file, /examples/PDQgraphicstest/Arduino_GFX_dev_device.h  shows how these are used with different displays. 
Try the example with various "types" selected to see what works for your display.

dagnall53, 2025-02-15:
Thom, thanks, the same Idea occurred to me just now, so I edited the original comment. I will try it and report the answers!

dagnall53, 2025-02-23:
Update. I found time to do some checks (with 1.3.8 as a start): (results below) were not totally encouraging..
But the latest update has a extra "type _init'- type9.. and also added some helpful comments in the init files:
they seem to allow set up for "IPS" - so I have modified for 'normal. in type1. and this works! 
For many versions.. but not 1.5.3..- I will post separate note) 

Change required is to set to NORMAL and not IPS.
    WRITE_COMMAND_8, 0x20,   // 0x20 normal, 0x21 IPS
    WRITE_C8_D8, 0x3A, 0x60, // 0x70 RGB888, 0x60 RGB666, 0x50 RGB565

The missing invertcolours function I would assume its technically possible to add one to the  'arduinoRGB_Display'
I will certainly try some hacking next. ((I think it requires sending 0x20 or 0x21 to location 0x0D.. ))

List of what the various initiation types showed.. just for reference! 
  type 1 is Inverted colours 
  type 2 is Inverted Colours very bright white washed out colours
  type 3 is NOT working ! just green with horizontal black line bars
  type 4   as 3 but orange (some flashing noticed when my display colours change)
  type 5   as 3
  type 6   as 3 black and white
  type 7   as 3 blue
  type 8  !! CORRECT COLOURS but I get some sort of artifact at bottom line of display.

moononournation, 2025-03-02:
You may try modify Arduino_RGB_Dispay.h line 511:
```
WRITE_COMMAND_8, 0x21,   // 0x20 normal, 0x21 IPS
```
Replace the value 0x21 by 0x20 to check is it all runs well.

If you can confirm this, I can duplicate a type1 for your display.

moononournation, 2025-03-02:
Or you may try type9 is it works well.

dagnall53, 2025-03-02:
Yes.. Thanks.
I should have sent earlier: 
I found the references in another ini, so changed the Write Commands  in st7701_type1_init_operations[] to 
    WRITE_COMMAND_8, 0x20,   // 0x20 normal, 0x21 IPS
    WRITE_C8_D8, 0x3A, 0x50, // 0x70 RGB888, 0x60 RGB666, 0x50 RGB565
A small comment. in some configurations, the RGB seems to reference the 18bit RGB's?
I think the boards  all use RGB565 ??  (16 pins) 
This works fine for my application. 
Many thanks

moononournation, 2025-03-02:
I think type9 should works for you, can you check this out?

dagnall53, 2025-03-02:
Yes, A quick test That seems to work as well. 
(Colours might be better than 'init1..)
Thanks. 

For anyone else having issues with the Guitron ESp32 4inch Touch screen : This is my settings:

[Esp32_4inch.zip](https://github.com/user-attachments/files/19043835/Esp32_4inch.zip)

moononournation, 2025-03-03:
Ok, thx you confirmed type9 can solve your problems.

closed 2025-03-03 by moononournation