moononournation/Arduino_GFX #628: Sunton 4827S043R board:  fb size must be even multiple of bounce buffer size

akomakom, 2025-02-22:
I have a `Sunton 4827S043R` board that works fine with Arduino_GFX (v8.3.0-dev) and ESP32 board support v2.x.   This is the version that came bundled from the seller/mfg, and seems to be some hash before 8.3.0.

I am trying to upgrade libs (ESP32 boards v2->v3, Arduino_GFX to latest - `v1.5.3` at the moment).

Any idea what could cause `fb size must be even multiple of bounce buffer size`?  I see this in the code:
```c++
.bounce_buffer_size_px = 40 * w,
```
Seems reasonable.

**Arduino version** is `1.8.19`

```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.01
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   384852 B ( 375.8 KB)
  Free Bytes        :   348956 B ( 340.8 KB)
  Allocated Bytes   :    30976 B (  30.2 KB)
  Minimum Free Bytes:   344228 B ( 336.2 KB)
  Largest Free Block:   278516 B ( 272.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8385672 B (8189.1 KB)
  Allocated Bytes   :      576 B (   0.6 KB)
  Minimum Free Bytes:  8385672 B (8189.1 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Feb 22 2025 06:54:12
  Compile Host OS   : linux
  ESP-IDF Version   : v5.3.2-584-g489d7a2b3a-dirty
  Arduino Version   : 3.1.3
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32S3_DEV
  Arduino Variant   : esp32s3
  Arduino FQBN      : esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=opi,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=debug,EraseFlash=all,ZigbeeMode=default
============ Before Setup End ============
Arduino_GFX Hello World example
E (361) lcd_panel.rgb: esp_lcd_new_rgb_panel(290): fb size must be even multiple of bounce buffer size
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x420034fc
file: "/home/akom/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_ESP32RGBPanel.cpp" line 134
func: uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)
expression: esp_lcd_new_rgb_panel(&panel_config, &_panel_handle)

abort() was called at PC 0x4037d137 on core 1

Backtrace: 0x4037664e:0x3fca4000 0x4037d141:0x3fca4020 0x40383b9d:0x3fca4040 0x4037d137:0x3fca40c0 0x420034fc:0x3fca40f0 0x420035ea:0x3fca41c0 0x4200236d:0x3fca41e0 0x420088df:0x3fca4210 0x4037dd5a:0x3fca4230

ELF file SHA256: 7ea660a2d

```

Here is the code:

```c++
/*
  Adapted from the Adafruit and Xark's PDQ graphicstest sketch.

  See end of file for original header text and MIT license info.
*/

/*******************************************************************************
 * Start of Arduino_GFX setting
 ******************************************************************************/
/* Arduino_GFX try to find the settings depends on selected board in Arduino IDE */
/* Or you can define the dev kit cannot find in board list */
#include <Arduino_GFX_Library.h>

#define TFT_BL 2

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
0 /* hsync_polarity */, 1 /* hsync_front_porch */, 1 /* hsync_pulse_width */, 43 /* hsync_back_porch */,
0 /* vsync_polarity */, 3 /* vsync_front_porch */, 1 /* vsync_pulse_width */, 12 /* vsync_back_porch */,
1 /* pclk_active_neg */, 8000000 /* prefer_speed */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
480 /* width */, 272 /* height */, rgbpanel, 0 /* rotation */, false /* auto_flush */);

/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{

  Serial.begin(115200);
  // Serial.setDebugOutput(true);
  // while(!Serial);
  Serial.println("Arduino_GFX Hello World example");

  // Init Display
  if (!gfx->begin())
  {
    Serial.println("gfx->begin() failed!");
  }
  gfx->fillScreen(RGB565_BLACK);

#ifdef TFT_BL
  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, HIGH);
#endif

  gfx->setCursor(10, 10);
  gfx->setTextColor(RGB565_RED);
  gfx->println("Hello World!");

  delay(5000); // 5 seconds
}

void loop()
{
  gfx->setCursor(random(gfx->width()), random(gfx->height()));
  gfx->setTextColor(random(0xffff), random(0xffff));
  gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
  gfx->println("Hello World!");

  delay(1000); // 1 second
}
```

moononournation, 2025-02-22:
You just want to ask a ESP32 related question, but not to raise a Arduino_GFX issue, right? Sorry, this is not the appropriate place to answer this question.

closed 2025-02-22 by moononournation

akomakom, 2025-02-22:
I have no idea if it is or if it isn't an Arduino_GFX issue. 

What I can say is that I get the issue above with 1.5.3, but not with 1.5.0.    
(Display isn't quite right with 1.5.0, but at least it isn't crashing).