moononournation/Arduino_GFX #630: ST7789 can't display correctly if display Height is bigger than 255

bmorcelli, 2025-02-22:
Hello,

I am [porting my project from TFT_eSPI to Arduino_GFX](https://github.com/bmorcelli/M5Stick-Launcher/tree/Arduino_gfx) and I'm facing problems with ST7789 displays.

When I set Height to 320 (default for this display) it overflows, and the display height become something near 65px.

If I set it to 255 (a full byte, 0xFF) it displays fine, but missing the other 65px 

I'm doing these tests on T-Deck, T-Deck Plus and T-Embed CC1101 and the issues are the same for both devices.. 

I've already ported to ESP-2432S028 and ESP-2432S028 in 240x320, which use ILI9341 display, and M5Core, Core2 and CoreS3, that uses ILI 9342 displays, and they are all working fine..

```
// T-Deck case
Arduino_DataBus *bus = new Arduino_HWSPI(11, 12, 40, 41, 38); // Using Arduino HWSPI to use the shared SPI with SDCard, but the issue happens when using ESP32SPI too
Arduino_GFX *tft = new Arduino_ST7789(bus, GFX_NOT_DEFINED/* RST */, 1 /* Rotation */, true /* IPS */, 240 /* Width */, 320 /* Height */, 0, 0, 0, 0);
```
Output:
[image]

```
// T-Deck case
Arduino_DataBus *bus = new Arduino_HWSPI(11, 12, 40, 41, 38); // Using Arduino HWSPI to use the shared SPI with SDCard, but the issue happens when using ESP32SPI too
Arduino_GFX *tft = new Arduino_ST7789(bus, GFX_NOT_DEFINED/* RST */, 1 /* Rotation */, true /* IPS */, 240 /* Width */, 255 /* Height */, 0, 0, 0, 0);
```
Output
[image]

Same happens with T-Embed
```
// T-Embed CC1101 case
Arduino_DataBus *bus = new Arduino_HWSPI(11, 12, 40, 41, 38); // Using Arduino HWSPI to use the shared SPI with SDCard, but the issue happens when using ESP32SPI too
Arduino_GFX *tft = new Arduino_ST7789(bus, GFX_NOT_DEFINED/* RST */, 1 /* Rotation */, true /* IPS */, 170 /* Width */, 320 /* Height */, 0, 0, 0, 0);
```
Output
[image]

```
// T-Embed CC1101 case
Arduino_DataBus *bus = new Arduino_HWSPI(11, 12, 40, 41, 38); // Using Arduino HWSPI to use the shared SPI with SDCard, but the issue happens when using ESP32SPI too
Arduino_GFX *tft = new Arduino_ST7789(bus, GFX_NOT_DEFINED/* RST */, 1 /* Rotation */, true /* IPS */, 170 /* Width */, 255 /* Height */, 0, 0, 0, 0);
```
Output

[image]

moononournation, 2025-02-22:
If you can run the PDQgraphicstest display correct, then it just your porting code issue.

closed 2025-02-22 by moononournation

bmorcelli, 2025-02-22:
You are right... 

Just for reference.. if using Arduino_HWSPI, must make sure to create an instance using HSPI.

Thank you!