moononournation/Arduino_GFX #631: Install Update via Arduino issue?  testing Version 1.5.3  with Arduino_RGB_Display

dagnall53, 2025-02-23:
Testing with the Cheap 4" display and st7701_type1_init_operations (see my other issue..)  
I was unable to compile with version 1.5.3  due apparently to a new flag: 
_error: 'esp_lcd_rgb_panel_config_t::<unnamed struct>' has no non-static data member named 'refresh_on_demand'_
**NOTE : I had updated my Arduino with the GFX 1.5.3 from the arduino Libraries selection.**

Delving deep. and looking at the github, I see you may have addressed this issue: and have also discovered that the Github "V1.5.3" version is NOT the same, as the version that gets installed by arduino as 1.5.3  !!

Git hub has the corrected updated Arduino_RGB_Display  ! and the critical Arduino_ESP32RGBPanel.cpp  ! 

I have tested by doing a simple Git zip extraction and pasting into my arduino libraries.. And now I have the changes.. And I can confirm it compiles.. 
and gives Inverted colours .. So I need to modify  WRITE_COMMAND_8, 0x20,   // 0x20 normal, 0x21 IPS in st7701_type1_init_operations
(not sure why you would have "inverted colours as the default!) ;-) 

BUT Why did these corrections not update when I updated the library via Arduino ?? 

do you need to Publish  V1.5.4 ?? 

 Many thanks for the Library!!

Loetnase, 2025-02-26:
Since 1.5.3 i have Problems too, with the RGB_Display

I use a **Sunton [ESP32-8048S070]**     (https://www.aliexpress.com/item/1005004952726089.html)
with the **Arduino_GFX/wiki/Dev-Device-Declaration**. 

It is running perfect with the 1.5.2 but never with 1.5.3 and 1.5.4 updatet with the Arduino IDE 2.3.2

I get the following warnings in red and some points or stripes on the TFT.

d:\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp: In member function 'uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)':
d:\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:75:35: warning: narrowing conversion of '(40 * ((int)w))' from 'int' to 'size_t' {aka 'unsigned int'} [-Wnarrowing]
   75 |       .bounce_buffer_size_px = 40 * w,
      |                                ~~~^~~

That is the Picture from the reboot

[image]

ist that the same Problem like in this issue or should i open an other new issue.

thanks for your help!

moononournation, 2025-02-26:
Please update 1.5.4

Loetnase, 2025-02-26:
Thank you for the fast answer,

but i had tested like i wrote with the 1.5.3 and the 1.5.4 with the same problem.
After your answer i updatet with the Arduino IDE **again to the 1.5.4** and have again **the same warning and problem**.

d:\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp: In member function 'uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)':
d:\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:75:35: warning: narrowing conversion of '(40 * ((int)w))' from 'int' to 'size_t' {aka 'unsigned int'} [-Wnarrowing]
   75 |       .bounce_buffer_size_px = 40 * w,
      |                                ~~~^~~

I'm sorry to write that, but it is like it is

dagnall53, 2025-02-27:
@Loetnase 
Can I suggest trying to download the zip and then loading the library from that zip in Arduino?
I discovered that for some reason, the 1.5.3 version on the github is slightly different from the version that Arduino Loads if you ask for '1.5.3' ..  (I used AgentRansack to check the differences. I can recommend it!) 
Please note: I have NOT yet tried updating from 1.5.3 to 1.5.4 via the arduino libraries update  ! (but I will when I can!..)

ThomDyson, 2025-02-27:
@Loetnase What version of board definition are you using? That line of code should only kick in if you are using older versions.

Loetnase, 2025-02-28:
@moononournation 
@ThomDyson i had first used the ESP32 3.1.2 and tested with the GFX 1.5.3 and 1.5.4
@dagnall53 installed first with the IDE and then with the ZIP from the Git    
then the same with the newest ESP32 3.1.3 and tested with the GFX 1.5.3 and 1.5.4 
first with the IDE and then with the ZIP from the Git again, nothing was working.

Then ESP32 3.1.3 and the GFX 1.5.2 where working again.

I use there the Sunton [ESP32-8048S070] with the ESP32S3-WROOM-1  N16R8 Witch is using PSRAM: OPI PSRAM
 

[image]

[image]

The Display shows with the mistake only stripes.
[image]

If i can help anything, tell me with wath.

thank you

ThomDyson, 2025-02-28:
From their demo file your code should look like this

```
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);
// option 1:
// 7寸 50PIN 800*480
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  bus,
//  800 /* width */, 0 /* hsync_polarity */, 8/* hsync_front_porch */, 2 /* hsync_pulse_width */, 43/* hsync_back_porch */,
//  480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 2/* vsync_pulse_width */, 12 /* vsync_back_porch */,
//  1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);

    800 /* width */, 0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    480 /* height */, 0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);
```
I see two options for *gfx settings. Did you try both?

Loetnase, 2025-02-28:
I use at the moment this one from the Wiki and it was working until 1.5.2
https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration#esp32-8048s070

but in your demo file the last three lines they look interesting with the timeing.
Where is this demofile to found?

In the LovyanGFX have i seen something simular for this Display

```
/* Start of Arduino_GFX setting for 7.0 Display */
#define GFX_BL 2  // GPIO 2 für Hintergrundbeleuchtung TFT Anzeige 7.0

Arduino_ESP32RGBPanel* rgbpanel = new Arduino_ESP32RGBPanel(
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
  9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
  0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
  0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display* gfx = new Arduino_RGB_Display(
  800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);
/* End of Arduino_GFX setting  */
```

ThomDyson, 2025-03-01:
The demo I found from the aliexpre link you posted. That had a link to manufacturer's  site.
to be clear, you are saying this worked 
https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration#esp32-8048s070
when you used Arduino GFX 1.5.2 with esp32 3.1.3  but did not work when you used 1.5.4?

Loetnase, 2025-03-01:
@ThomDyson 
yes, i have this board arround 2 years and playing wit it. sometimes.
It was working the whole time with different GFX versions up to 1.5.2, then last week i datet up to the GFX 1.5.3 and noting was working, i found this issue and wrote then i datet up to the GFX 1.5.4 and made all what i wrote.

Now i'm back to  GFX 1.5.2 and it works again.

moononournation, 2025-03-01:
Yes, you also require update the arduino_esp32 board firmware as well while update Arduino_GFX.

closed 2025-03-01 by moononournation

Loetnase, 2025-03-01:
@moononournation 
I'm sorry i cant reopen the issue, but it is still the same 

Do yo mean the Arduino_esp32?  this newest board firmware is installed too.
 
I use the plain newest ESP32 from Espessif, the Sunton  is not a Arduino Board 

[image]

i get this warning during compiling and the GFX 1.5.4 is not working

d:\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp: In member function 'uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)':

d:User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp:75:35: warning: narrowing conversion of '(40 * ((int)w))' from 'int' to 'size_t' {aka 'unsigned int'} [-Wnarrowing]
   75 |       .bounce_buffer_size_px = 40 * w,
      |                                ~~~^~~

And back to GFX 1.5.2, i get no warning during compiling and the Display with the GFX 1.5.2 is working again.

Thank you for your work for us all

dagnall53, 2025-03-01:
@Loetnase 
that error is the same I got when I compiled with ESP32 by expressive V2.0.11 and the GFX library 1.5.3 as uploaded by Arduino.
However, when I compared the '1.5.3 as shown on github, it had extra definitions included to make it compatible with the new Esp32 compilers version 3, (that have a lot of what are nasty changes for programmers). 
I think @moononournation re- 'uploaded' with the new version number 1.5.4, but I have not yet had time to load that version to check it includes the crucial elements around those warnings, that concern elements he had dded to make the library work with Version 3 compiler. (but to try and keep backwards compatibility (!thank you !) .
Can I suggest again that if you want to try V1.5.4 (or 1.5.3), you download the zip from the github and install the library by Include library ...add from zip.  These seems to overcome the issue with Arduino not uploading the correct version.

moononournation, 2025-03-02:
You should install latest arduino_esp32 in Arduino IDE Board Manager, current version should be 3.1.3.