moononournation/Arduino_GFX #633: Add support for rp2350

3shulmotors-dev, 2025-02-27:
Hey first of all thanks for this great work. 

I have got the 4d systems rp2350 based 5inch rgb 16bit display and find no library that supports it. 

Are there any future plans to add support for this microcontroller.

ThomDyson, 2025-02-27:
The 2350 looks like a Raspberry Pi device. This library is for arduino devices.

3shulmotors-dev, 2025-02-27:
> The 2350 looks like a Raspberry Pi device. This library is for arduino devices.

Yes but this library support arduino platform compatible mcu like esp32 and this mcu is also arduino compatible and very capable for display driving applications.

ThomDyson, 2025-02-27:
I just checked the repo here.  There are mentions that 2350 support was added last month.
https://github.com/moononournation/Arduino_GFX/tree/master/src/databus

3shulmotors-dev, 2025-02-27:
Yes but those are for 8080mcu 8 and 16 bit parallel interfaces and not for RGB 16bit interface.

ThomDyson, 2025-02-27:
Is there any lib that supports 16bit that under arduino?

3shulmotors-dev, 2025-02-27:
For this particular chip no. But there is 4d systems library that is supporting rgb on this display but doesn't support arduino and lvgl. But yes there are a few library like this one and lovyan gfx that support RGB 16bit interface on esp32 with arduino platform.

ThomDyson, 2025-02-27:
I'm a bit out of my depth here, but if you can use that interface, then you just need the settings, right? The pin outs are here, https://resources.4dsystems.com.au/datasheets/rp2350/pdf/gen4-RP2350-RGB-REV1.2.pdf, but you still have to figure out hsync, vsyc etc. Can you pull those values from the 4D example code?

3shulmotors-dev, 2025-02-28:
> I'm a bit out of my depth here, but if you can use that interface, then you just need the settings, right? The pin outs are here, [...]

Yes but first rp2350rgb support need to be added in this library to use this we cant just use any other mcu driver on rp2350 also yes 4d systems have full code that will help with adding support to this librarry.

I will upload the code later today.

moononournation, 2025-02-28:
Please do not raise an issue for a new hardware support request.

closed 2025-02-28 by moononournation