moononournation/Arduino_GFX #637: Problems since after version 1.5.1

CoolmdXi, 2025-03-08:
I have previously been using your library with no problems on dev board JC8048W550. I was using this setup and never had an issue.

#define GFX_DEV_DEVICE ESP32_4827S043
#define GFX_BL 2
#define RGB_PANEL
// Uncomment for ST7262 IPS LCD 800x480
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
  45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
  5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
  8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
  0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
  0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
  1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
  800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

since update of version 1.5.1 the display is no longer correct, lots of noise and flicker and does not look like display is being writtain at correct coordinates. But hard to say other than very unstable and flickers very badly. If i take the version back to 1.5.1 everything is perfect
Regards
Mark D

moononournation, 2025-03-09:
try lower the prefer_speed to 12000000

CoolmdXi, 2025-03-09:
I did try the speed from 12000000 in increments of 500000, and although it helped, it never became very stable.
I will retry some more later, and will let you know if I find a sweet spot.

CoolmdXi, 2025-03-09:
I tried many variations to no avail. If i use your clock sketch, the clock face is drawn overlapping top of display and wrapping onto bottom of display. With the PDQ test, lots of strange results including report at end of sketch being sometimes split and incorrectly displayed.
I tried many different speeds between 12 which gives a white screen and 16000000 but nothing would correct the errors. i tried rolling back and the last fully working version appears to be 1.5.1.
I am happy to use that but thought i would let you know.
I also ran some tests on the JC4827W543 and can say that this seems to run perfectly on the latest updates.
Many thanks.

ThomDyson, 2025-03-11:
What version of the esp32 board definition do you have?

AndreyUshak, 2025-03-13:
Hi, I have an ESP32S3-8048S070. 
 on ESP 32 v3.1.3, it just flickers,
on 2.0.9, 2.0.17 it works correctly.
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
  9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
  //  0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
  //  0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);

  0 /* hsync_polarity */, 240 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
  0 /* vsync_polarity */, 32 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10, /* vsync_back_porch */
  1 /* pclk_active_neg */, 12000000 /* prefer_speed */);

ThomDyson, 2025-03-14:
> Hi, I have an ESP32S3-8048S070. on ESP 32 v3.1.3, it just flickers, 

Try ESP32 v 3.1.1

moononournation, 2025-03-22:
Please get update to v1.5.6 and follow latest example to test again.

closed 2025-03-22 by moononournation