moononournation/Arduino_GFX #638: Commit 9df9aeb breaks Waveshare 4.3 RGB display support

ThomDyson, 2025-03-09:
This change,
`from:      .bounce_buffer_size_px = 0,
to:   .bounce_buffer_size_px = 40 * w,`

causes a crash with complex screen updates and other DMA actions.  I know v1.5.5 is slightly different, but the problem started here.  This configuration

```
Arduino_ESP32RGBPanel *myRGBpanel = new Arduino_ESP32RGBPanel(
  5 /* DE */, 3 /* VSYNC */, 46 /* HSYNC */, 7 /* PCLK */, 1 /* R0 */,
  2 /* R1 */, 42 /* R2 */, 41 /* R3 */, 40 /* R4 */, 39 /* G0 */, 0 /* G1 */,
  45 /* G2 */, 48 /* G3 */, 47 /* G4 */, 21 /* G5 */, 14 /* B0 */,
  38 /* B1 */, 18 /* B2 */, 17 /* B3 */, 10 /* B4 */,

  0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
  0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
  1 /* pclk_active_neg */, 14000000 /* prefer_speed */

);

Arduino_RGB_Display *gfx =
  new Arduino_RGB_Display(screenWidth /* width */, screenHeight /* height */,
                          myRGBpanel, 0 /* rotation */, true /* auto_flush */
  );
```

 works prior to this commit. I've changed that line back and forth to confirm it is the issue.  Updating to v 1.5.5 did not solve the problem.  Rolling back just that one line in v 1.5.5 did solve the crashing.

The PDQ test works, but that doesn't stress the system much and has no other DMA actions.

My sketch cashes with this exception stack
```
Decoding stack results
0x403795cc: lcd_rgb_panel_fill_bounce_buffer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_lcd/rgb\esp_lcd_panel_rgb.c:929
0x4037978b: lcd_rgb_panel_eof_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_lcd/rgb\esp_lcd_panel_rgb.c:964
0x4037bd8b: gdma_default_tx_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/dma\gdma.c:851
0x403763a8: spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash\cache_utils.c:134
0x4037662f: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system\esp_ipc.c:65
0x4037fa0e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/Free

```

This also causes a crash with a simple test sketch using LittleFS and just a basic GFX init and screen draw. I traced that to the same line in esp-lcd_panel_rgb.  It looks like some sort of DMA conflict.

Board definitions are v3.1.3

moononournation, 2025-03-10:
I think it should report to arduino-esp32 repository

ThomDyson, 2025-03-10:
Can confirm. GFX v1.5.5 with board definitions 3.1.1 works for both of my test cases.   Thanks.

closed 2025-03-10 by ThomDyson