moononournation/Arduino_GFX #641: Flickering issue with ESP32-8048S070

docentYT, 2025-03-16:
Problem: https://i.imgur.com/rIOwkUw.mp4

My code:
```cpp
#include <Arduino.h>
#include <Arduino_GFX_Library.h>
#define GFX_BL 2

#define TFT_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */, 14 /* R0 */,
    21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */, 9 /* G0 */, 46 /* G1 */,
    3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */, 15 /* B0 */, 7 /* B1 */,
    6 /* B2 */, 5 /* B3 */, 4 /* B4 */, 0 /* hsync_polarity */,
    180 /* hsync_front_porch */, 30 /* hsync_pulse_width */,
    16 /* hsync_back_porch */, 0 /* vsync_polarity */,
    12 /* vsync_front_porch */, 13 /* vsync_pulse_width */,
    10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx =
    new Arduino_RGB_Display(800 /* width */, 480 /* height */, rgbpanel,
                            0 /* rotation */, true /* auto_flush */);

void setup() {
    gfx->begin();

#ifdef TFT_BL
    pinMode(TFT_BL, OUTPUT);
    digitalWrite(TFT_BL, HIGH);
#endif

    gfx->fillScreen(BLACK);
}

void loop() {}
```

If I run only 
```cpp
pinMode(TFT_BL, OUTPUT);
digitalWrite(TFT_BL, HIGH);
```
without Arduino_GFX library it does not flicker.

ThomDyson, 2025-03-16:
Try it with the esp32 version 3..1.1  under board manager.

docentYT, 2025-03-16:
> Try it with the esp32 version 3..1.1 under board manager.

I forgot to mention that I am using platformio. I've installed Arduino IDE with esp32 version 3.1.3 and the problem still persists. Then I've downgraded to 3.1.1 and it does not resolved the problem.

docentYT, 2025-03-16:
The strange thing is that if you wait long enough with the program on, the screen flickers less and less.

ThomDyson, 2025-03-16:
The flickering is a timing issue. You'll have to google around for some example and try different settings I think.

docentYT, 2025-03-16:
Timing of what do you mean?

ThomDyson, 2025-03-17:
These settings control the timing and details of the signal to the display
` 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */,
    16 /* hsync_back_porch */, 0 /* vsync_polarity */,
    12 /* vsync_front_porch */, 13 /* vsync_pulse_width */,
    10 /* vsync_back_porch */);`

If you don't have sample code from the vendor, you can google your model number. I found this set of possible matches
`    0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    , 0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */)`

Also, it looks like you are missing a couple of parameters.

docentYT, 2025-03-17:
I've used values from [Wiki](https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration#esp32-8048s070):
```cpp
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);
```

From the vendor I've got values for older Arduino_GFX version:
```cpp
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);
// option 1:
// 7寸 50PIN 800*480
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  bus,
//  800 /* width */, 0 /* hsync_polarity */, 8/* hsync_front_porch */, 2 /* hsync_pulse_width */, 43/* hsync_back_porch */,
//  480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 2/* vsync_pulse_width */, 12 /* vsync_back_porch */,
//  1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);

    800 /* width */, 0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    480 /* height */, 0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);
```
This also causes flickering.

I've tried [values from LGFX library](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx_user/LGFX_Sunton_ESP32-8048S070.h), but the problem still exists. Although sometimes it happens that it does not flash, or flashes less. The same is true for the values you sent. 
However, using the LGFX library, I don't see any flickering, which would suggest that Arduino_GFX is causing some kind of bad clock accuracy (?).

docentYT, 2025-03-20:
No. I've switched to LovyanGFX. 

I am sending you our displays specification, so you can play with timings for Arduino_GFX if you want.

[JC8048B070N.pdf](https://github.com/user-attachments/files/19373700/JC8048B070N.pdf)

moononournation, 2025-03-20:
Yes, LovyanGFX is much better. Please make it a chance.

joaomarcelo-dev, 2025-03-21:
@docentYT, would you have a code example that worked perfectly on your board using the LovyanGFX library?

docentYT, 2025-03-21:
Sure
```cpp
#include <LovyanGFX.hpp>
#include <lgfx_user/LGFX_Sunton_ESP32-8048S070.h>

LGFX tft;

void setup()
{
  tft.begin();

  tft.setFont(&fonts::DejaVu40);
  tft.drawRect(0, 0, tft.width(), tft.height(), TFT_GREEN);
  tft.setTextColor(TFT_GOLD);
  tft.setCursor(30, 40);
  tft.print("Hello World!");
  tft.fillCircle(400, 150, 100, TFT_SKYBLUE);
}
```

If it flickers after uploading the program restart the screen a few times and this problem will not occur again.

joaomarcelo-dev, 2025-03-21:
Can you compile it for the ESP32S3 Dev Module?

docentYT, 2025-03-21:
In arduino IDE you have to set values as shown in the official display's instruction:
[Getting started 7.0 Inch.pdf](https://github.com/user-attachments/files/19395187/Getting.started.7.0.Inch.pdf)

I am using platformio instead of arduino ide and this is my platformio.ini:
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
board_build.arduino.memory_type = qio_opi
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.flash_mode = qio
board_build.f_flash = 80000000L
build_flags = 
	-DBOARD_HAS_PSRAM
monitor_speed = 115200
upload_speed = 921600
lib_deps = lovyan03/LovyanGFX@^1.2.0
```

joaomarcelo-dev, 2025-03-21:
@docentYT Thank you very much for your help! It worked perfectly on my board! Good luck with your projects!

closed 2025-03-22 by moononournation