moononournation/Arduino_GFX #647: Whole screen scrolls up without any request - not fixed using version 1.5.6

vitasynergy, 2025-03-23:
**The unsolicited whole displayed image scroll issue opened with #646 has not been fixed installing and using the GFX library version 1.5.6.**

The whole image scrolls also when it is used the following simple sequence of instructions:
        gfx->setCursor(......);
        gfx->print(" whatever text ");

or this:
        jpegDraw(imageFile, jpegDrawCallback, true, 
                  xCoord, yCoord, 
                  gfx->width(), gfx->height());

The scroll seems to happen less frequently only when the instructions being used are the following:
        gfx->drawCircle(.....);
        gfx->fillCircle(.....);

ThomDyson, 2025-03-23:
Please post your complete code and I'll try it on Monday. Without the jpg call if possible

vitasynergy, 2025-03-23:
Here it is the code. 

Please note that the code fully depends on the system state variable "statoSistema" (systemState) : if "INATTESA" (means IFWAYTING), then the system shows a background image (SFONDO_STARTUP.jpg) and alternatively two "waiting" images to show to the user a waiting state (waiting for something to happen..). The two images are WAIT_00.jpg and WAIT_01.jpg.
In such a INATTESA state, where a background image is drawn and two images are drawn to the display center, the scroll happens (sometimes immediately, sometimes with a delay): all the display scrolls up. 

[image]
[image]
[image]
[test_Arduino_GFX_library.txt](https://github.com/user-attachments/files/19410731/test_Arduino_GFX_library.txt)

moononournation, 2025-03-27:
Please do not provide foreign link here.

closed 2025-03-27 by moononournation

vitasynergy, 2025-03-27:
This is your way to reply to the issues ? Really incredible.

There was no "foreign link" in the code and images I've uploaded to allow you to fix the problem.

You took more than four days to say the absolute cosmic nothing.

Gartosz, 2025-04-07:
I bet it's a problem with your environment/configuration/screen, but in all your issues you didn't include necessary information like logs or at least board settings. We don't even know if you were able to compile at least a basic "Hello World" program.

ricercarfuga, 2025-04-08:
you're betting on the wrong horse.

In all my previous posts there was plenty of info, but obviously didn't have the desire or time to read them i.e.at post https://github.com/moononournation/Arduino_GFX/issues/646 .

Furthermore in my reply two weeks ago to Mr. "moononournation", I even provided ALL the source code of my app and in response Mr. "moononournation" deleted my post commenting "Please do not put foreign links here" (obviously there was no "foreign" link in the code).

In any case even after a simple fillRect() and flush() all the displayed content scrolls, a few pixels at a time, but scrolls.

vitasynergy, 2025-04-08:
you're betting on the wrong horse.
In all my previous posts there was plenty of info, but obviously didn't have the desire or time to read them such as post [#646](https://github.com/moononournation/Arduino_GFX/issues/646) .
Furthermore in my reply two weeks ago to Mr. "moononournation", I even provided ALL the source code of my app and in response Mr. "moononournation" deleted my post commenting "Please do not put foreign links here" (obviously there was no "foreign" link in the code).
In any case even after a simple fillRect() and flush() all the displayed content scrolls, a few pixels at a time, but scrolls.

ThomDyson, 2025-04-08:
It would have been better to just post a "hello World" minimal example. Nobody really wants to wade through a repo to find a specific problem.  I have the Waveshare 4.3 and I use this configuration

`  5 /* DE */,
  3 /* VSYNC */,
  46 /* HSYNC */,
  7 /* PCLK */,

  1 /* R0 */,
  2 /* R1 */,
  42 /* R2 */,
  41 /* R3 */,
  40 /* R4 */,

  39 /* G0 */,
  0 /* G1 */,
  45 /* G2 */,
  48 /* G3 */,
  47 /* G4 */,
  21 /* G5 */,

  14 /* B0 */,
  38 /* B1 */,
  18 /* B2 */,
  17 /* B3 */,
  10 /* B4 */,
  0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
  0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
  1 /* pclk_active_neg */, 14000000 /* prefer_speed */`

You linked to a "B" configuration wiki, but I use this as my reference https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3  I don't know if it matters.

Using library GFX Library for Arduino at version 1.5.6 
Using library SPI at version 3.1.1 
Using library Wire at version 3.1.1 
Using library bb_captouch at version 1.2.2 
Using library ESP32_IO_Expander at version 1.1.0 
Using library esp-lib-utils at version 0.2.0 
ESP32 board definition 3.1.1 - this is not the most current, but works

Gartosz, 2025-04-08:
> you're betting on the wrong horse. In all my previous posts there was plenty of info, but obviously didn't have the desire or time to read them such as post [#646](https://github.com/moononournation/Arduino_GFX/issues/646) .

I've read all 3 of your issues and in none there were your full code posted.

vitasynergy, 2025-04-09:
forgive me, but no, you didn't read (and, worse, you didn't understand what you read): the post in which i provided all the source of my app has been DELETED by moononournation.

vitasynergy, 2025-04-09:
an update to moononournation : using the Arduino_ESP32RGBPanel setting you're using there's no more screen scroll, even if my 4.3" waveshare display is slightly different from your (mine "B" model has all the connection placed on the display back) and the memory model is different having 16MB flash.
So, thank you very much indeed.

Gartosz, 2025-04-09:
> forgive me, but no, you didn't read (and, worse, you didn't understand what you read): the post in which i provided all the source of my app has been DELETED by moononournation.

I'm talking about this post, #646 [#93](https://github.com/bitbank2/JPEGDEC/issues/93) from JPEGDEC. In none of these is full code provided apart from classes initalizations just pasted as text (which doesn't look good). I can't say anything about what you've commented and it was deleted, so how could I see your source? You could just paste everything there like that (but with actual code).

`
Arduino_ESP32RGBPanel* rgbpanel = new Arduino_ESP32RGBPanel(
5 /* DE /, 3 / VSYNC /, 46 / HSYNC /, 7 / PCLK /,
1 / R0 /, 2 / R1 /, 42 / R2 /, 41 / R3 /, 40 / R4 /,
39 / G0 /, 0 / G1 /, 45 / G2 /, 48 / G3 /, 47 / G4 /, 21 / G5 /,
14 / B0 /, 38 / B1 /, 18 / B2 /, 17 / B3 /, 10 / B4 /,
0 / hsync_polarity /, 40 / hsync_front_porch /, 48 / hsync_pulse_width /, 88 / hsync_back_porch /,
0 / vsync_polarity /, 13 / vsync_front_porch /, 3 / vsync_pulse_width /, 32 / vsync_back_porch /,
1 / pclk_active_neg /, 16000000 / prefer_speed */
);

//void GFXdraw16bitBeRGBBitmap(int16_t x1, int16_t y1, uint16_t* full, int16_t w, int16_t h);

Arduino_RGB_Display* gfx = new Arduino_RGB_Display(
800 /* width /, 480 / height /, rgbpanel, 0 / rotation /, true / auto_flush */
);
`

Also it seems that it turned out, like I said before, that your issue was related to configuration as it works now with different settings.