moononournation/Arduino_GFX #649: Newer version of GFX_Library_for_Arduino does work with Waveshare ESP32-S3-Touch-AMOLED-1.8

richgain, 2025-03-28:
I recently bought a Waveshare ESP32-S3-Touch-AMOLED-1.8 and I was having great difficulty getting any of the example projects to compile. Compilation regularly stopped with the following error:
```C:\Users\richg\Documents\Arduino\Waveshare_ESP32-Amoled\examples\01_HelloWorld\01_HelloWorld.ino:14:24: error: expected type-specifier before 'Arduino_SH8601'
   14 | Arduino_GFX *gfx = new Arduino_SH8601(bus, -1 /* RST */,
      |                        ^~~~~~~~~~~~~~
C:\Users\richg\Documents\Arduino\Waveshare_ESP32-Amoled\examples\01_HelloWorld\01_HelloWorld.ino: In function 'void setup()':
C:\Users\richg\Documents\Arduino\Waveshare_ESP32-Amoled\examples\01_HelloWorld\01_HelloWorld.ino:32:8: error: 'class Arduino_GFX' has no member named 'Display_Brightness'
   32 |   gfx->Display_Brightness(255);
      |        ^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: expected type-specifier before 'Arduino_SH8601'
```

In the end, I archived my whole 'libraries' folder and started a fresh one with only the libraries provided in the examples zip folder. This immediately fixed the problem and allowed me to try out the examples.

Then I tried updating the GFX_Library_for_Arduino from 1.4.9 to the latest version 1.5.6 and the same error came back again.
Strangely, I found that reverting the GFX library to version 1.4.9 did not fix the problem, but deleting the folder and copying back the one from the zip archive did fix it.

Finally, I restored my original Arduino library archive, deleted the GFX_Library_for_Arduino folder and copied the 1.4.9 version, and once again everything is compiling correctly.

moononournation, 2025-03-29:
Sorry, I did not have this hardware in hand. That's why you cannot find it in the supported Dev Device list.

closed 2025-03-29 by moononournation

razarahil, 2025-04-08:
Same Issue, the latest version gives the compilation error.

richgain, 2025-04-27:
Will the ESP32-S3 1.8inch AMOLED Touch be a supported device in the future? I love this little thing!  😁

moononournation, 2025-04-28:
I do not have this in hand, but welcome you contribute a pull request.