moononournation/Arduino_GFX #654: Crash esp32-S3 with ESP32 library v3.2.0

lebossejames, 2025-04-08:
Dear,

I have a crash, the esp32 reboot automatically, esp32-S3 with ESP32 library v3.2.0 and GFX library:

`#include <U8g2lib.h>
#include <Arduino_GFX_Library.h>

/* OPTION 1: Uncomment a dev device in Arduino_GFX_dev_device.h */
#include "Arduino_GFX_dev_device.h"

#include <SD.h>
#include <FS.h>
#include <SPI.h>
#include <ArduinoJson.h>
#include <WiFi.h>
#include "time.h"

int nbFails=0;
const char* ssid     = "test";
const char* password = "test";
bool isWifiConnected=false;

void setup()
{
  Serial.begin(115200);
  Serial.setDebugOutput(false);
  
  TaskHandle_t Task_Second;
  TaskHandle_t Task_Main;
xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task_Main",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    5,           /* priority of the task */
                    &Task_Main,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  
   
      delay(500);
      //create a task that will be executed in the Task2code() function, with priority 0 and executed on core 0
      xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task_Second",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task_Second,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 0 */
 
      delay(500);

}

void Task1code( void * pvParameters ){
  Serial.println("Task1 running on core "+xPortGetCoreID());

  for(;;){

    ManageWebAndDisplay();

 
  }  // for
}

void Task2code( void * pvParameters ){
 Serial.println("Task2 running on core "+xPortGetCoreID());

  for(;;){
    
  
      Manage();    
  }
}

void ManageWebAndDisplay(){
  Serial.println(F("ManageWebAndDisplay"));

  if (isWifiConnected == false){ 
    Serial.println(isWifiConnected);
    connectToAP();
  }
  
  int32_t usecFilledRoundRects=displayBlockHours();
  serialOut(F("Rounded rects (filled)\t"), usecFilledRoundRects, 100, true);
  delay(2000);
  
  }
  
  
  void Manage(){
  delay(1000);
  }
  
  bool connectToAP(){

// Connect to Wi-Fi
  Serial.print("Connecting to ");
  Serial.println(ssid);
  //WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  //delay(1000);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    nbFails++;
    Serial.print(".");
    if(nbFails>=20){
        nbFails=0;
        isWifiConnected=false;
        break;
    }
  }
  if (WiFi.status() == WL_CONNECTED){ 
    Serial.println("");
    Serial.println("WiFi connected.");
    isWifiConnected=true;
    return true;
  }else{
    // unconnected
    isWifiConnected=false;
    WiFi.disconnect(true);
    WiFi.mode(WIFI_OFF);
    return false;
  }

  //disconnect WiFi as it's no longer needed
  //WiFi.disconnect(true);
  //WiFi.mode(WIFI_OFF);

}
int32_t displayBlockHours()
{
  uint32_t start;
  uint16_t colorGreen=gfx->color565(1, 31 , 9);
start = micros_start();

  gfx->setCursor(0, 10);

  gfx->setTextSize(1);
  gfx->setTextColor(WHITE, BLACK);
  
  gfx->println(F("Title"));

return micros() - start;
}  
`

`ESP Exception Decoder
Sketch: MyTest FQBN: esp32:esp32:esp32s3

A6      : 0x3fca9624  A7      : 0x420be8b0  A8      : 0x00002580  A9      : 0x00000000  
A10     : 0x3fc9cab0  A11     : 0x00000050  A12     : 0x3fcb938c  A13     : 0xffffffff  
A14     : 0x00000005  A15     : 0x3fc9c950  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00002580  LBEG    : 0x400570a4  LEND    : 0x400570a9  LCOUNT  : 0x00000000  

Backtrace: 0x420be8d9:0x3fcb9230 0x42006ecd:0x3fcb9250 0x42006a05:0x3fcb9290 0x420bf023:0x3fcb92c0 0x420bf046:0x3fcb92e0 0x42013056:0x3fcb9300 0x42003cd4:0x3fcb9320 0x420044a7:0x3fcb9350 0x4200454c:0x3fcb93b0 0x4037ef42:0x3fcb93d0

EXCVADDR: 0x00002580

Decoding stack results
0x420be8d9:  is in Arduino_Canvas::writePixelPreclipped(short, short, unsigned short) (/home/james/Arduino/libraries/GFX_Library_for_Arduino/src/canvas/Arduino_Canvas.cpp:76).
0x42006ecd:  is in Arduino_GFX::drawChar(short, short, unsigned char, unsigned short, unsigned short) (/home/james/Arduino/libraries/GFX_Library_for_Arduino/src/Arduino_GFX.cpp:2217).
0x42006a05:  is in Arduino_GFX::write(unsigned char) (/home/james/Arduino/libraries/GFX_Library_for_Arduino/src/Arduino_GFX.cpp:2485).
0x420bf023:  is in Print::write(unsigned char const*, unsigned int) (/home/james/.arduino15/packages/esp32/hardware/esp32/3.2.0/cores/esp32/Print.cpp:41).
0x420bf046:  is in Print::print(String const&) (/home/james/.arduino15/packages/esp32/hardware/esp32/3.2.0/cores/esp32/Print.h:66).
0x42013056:  is in Print::println(String const&) (/home/james/.arduino15/packages/esp32/hardware/esp32/3.2.0/cores/esp32/Print.cpp:174).
0x42003cd4: displayBlockHours() at /home/james/Arduino/MyTest/MyTest.ino:1343
0x420044a7: ManageWebAndDisplay() at /home/james/Arduino/MyTest/MyTest.ino:473
0x4200454c: Task1code(void*) at /home/james/Arduino/MyTest/MyTest.ino:329
0x4037ef42: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

`

moononournation, 2025-04-08:
can you run PDQgraphicstest normally?

lebossejames, 2025-04-09:
On ESP32 forum, its notify as a GFX issue: https://esp32.com//viewtopic.php?p=146473#p146473 as i see on most forums.

moononournation, 2025-04-09:
if PDQgraphicstest run ok, then you need to fix the code you add

closed 2025-04-14 by moononournation