moononournation/Arduino_GFX #655: Buffer sizes

frankcohen, 2025-04-09:
I am seeking clarification on the memory needs of JPEGDEC and Arduino_GFX. [ImgViewerMjpeg_JPEGDEC](https://github.com/moononournation/Arduino_GFX/blob/master/examples/ImgViewer/ImgViewerMjpeg_JPEGDEC/ImgViewerMjpeg_JPEGDEC.ino) example sets the memory buffer using: 

```
#define MJPEG_BUFFER_SIZE (240 * 240 * 2 / 10) // memory for a single JPEG frame
```

[ImgViewerMjpeg_ESP32P4](https://github.com/moononournation/Arduino_GFX/tree/master/examples/ImgViewer/ImgViewerMjpeg_ESP32P4) example uses this:

```
#define MJPEG_OUTPUT_SIZE (240 * 240 * 2)          // memory for a output image frame
#define MJPEG_BUFFER_SIZE (MJPEG_OUTPUT_SIZE / 10) // memory for a single JPEG frame
```

And the [Canvas](https://github.com/moononournation/Arduino_GFX/blob/master/src/canvas/Arduino_Canvas.cpp) for double buffering  uses this:

```
size_t s = _width * _height * 2;
#if defined(ESP32)
    _framebuffer = (uint16_t *)aligned_alloc(16, s);
```

I'm hunting around my app to free up memory. What does dividing the allocation size by 10 do? Could I get away with a smaller allocated memory block?

-Frank

moononournation, 2025-04-09:
240x240 is the image dimension and the 10 factor is the estimated JPEG compression ratio. Hope this can help you.

closed 2025-04-09 by moononournation

frankcohen, 2025-04-09:
Thanks that helps. So I could go for 20 provided the compression ratio isn't larger. What happens when it's larger?