moononournation/Arduino_GFX #660: 能否 添加 HX8394 5.5寸1280*720 MIPI屏驱动

Big-mouth-c, 2025-04-17:
您好：
    我再使用 正点原子的 ESP32P4时使用了5.5寸 1280*720 MIPI屏。发现这个库当前不支持HX8493驱动
   我在 Arduino_DSI_Display.h 添加了下代码，现在使用您的这个库GFX库 已经能够正常显示 HellWord示例了。
  同时在 Arduino_ESP32DSIPanel.cpp 中需要修改 .lane_bit_rate_mbps = 700 。原库设定750会在屏幕上出现水波纹。

示例代码 显示屏 设置代码段
#include <Arduino_GFX_Library.h>

#define GFX_BL 53 // default backlight pin, you may replace DF_GFX_BL to actual backlight pin
#define DSI_PANEL
Arduino_ESP32DSIPanel *dsipanel = new Arduino_ESP32DSIPanel(
  8 /* hsync_pulse_width */, 52 /* hsync_back_porch */, 48 /* hsync_front_porch */,
  5 /* vsync_pulse_width */, 15 /*vsync_back_porch  */, 16 /* vsync_front_porch */,
  60000000 /* prefer_speed */);
Arduino_DSI_Display *gfx = new Arduino_DSI_Display(
  720 /* width */, 1280 /* height */, dsipanel, 1 /* rotation */, true /* auto_flush */,
  52 /* RST */, hx8394_init_operations, sizeof(hx8394_init_operations) / sizeof(lcd_init_cmd_t));

Arduino_DSI_Display.h  HX8394 驱动代码段

/* HX83xx User Define command set 用户自定义命令集 */
#define UD_SETADDRESSMODE                       0x36    /* Set address mode */
#define UD_SETSEQUENCE                          0xB0    /* Set sequence */
#define UD_SETPOWER                             0xB1    /* Set power */
#define UD_SETDISP                              0xB2    /* Set display related register */
#define UD_SETCYC                               0xB4    /* Set display waveform cycles */
#define UD_SETVCOM                              0xB6    /* Set VCOM voltage */
#define UD_SETTE                                0xB7    /* Set internal TE function */
#define UD_SETSENSOR                            0xB8    /* Set temperature sensor */
#define UD_SETEXTC                              0xB9    /* Set extension command */
#define UD_SETMIPI                              0xBA    /* Set MIPI control */
#define UD_SETOTP                               0xBB    /* Set OTP */
#define UD_SETREGBANK                           0xBD    /* Set register bank */
#define UD_SETDGCLUT                            0xC1    /* Set DGC LUT */
#define UD_SETID                                0xC3    /* Set ID */
#define UD_SETDDB                               0xC4    /* Set DDB */
#define UD_SETCABC                              0xC9    /* Set CABC control */
#define UD_SETCABCGAIN                          0xCA
#define UD_SETPANEL                             0xCC
#define UD_SETOFFSET                            0xD2
#define UD_SETGIP0                              0xD3    /* Set GIP Option0 */
#define UD_SETGIP1                              0xD5    /* Set GIP Option1 */
#define UD_SETGIP2                              0xD6    /* Set GIP Option2 */
#define UD_SETGIP3                              0xD8    /* Set GIP Option2 */
#define UD_SETGPO                               0xD9
#define UD_SETSCALING                           0xDD
#define UD_SETIDLE                              0xDF
#define UD_SETGAMMA                             0xE0    /* Set gamma curve related setting */
#define UD_SETCHEMODE_DYN                       0xE4
#define UD_SETCHE                               0xE5
#define UD_SETCESEL                             0xE6    /* Enable color enhance */
#define UD_SET_SP_CMD                           0xE9
#define UD_SETREADINDEX                         0xFE    /* Set SPI Read Index */
#define UD_GETSPIREAD                           0xFF    /* SPI Read Command Data */

static const lcd_init_cmd_t hx8394_init_operations[] = {
    //  {cmd, { data }, data_size, delay_ms}
    /* 720 * 1280 */
    {UD_SETADDRESSMODE, (uint8_t []){0x01}, 1,0},
    {UD_SETEXTC,        (uint8_t []){0xFF, 0x83, 0x94}, 3,0},
    {UD_SETMIPI,        (uint8_t []){0x61, 0x03, 0x68, 0x6B, 0xB2, 0xC0}, 6,0}, //0x61
    {UD_SETPOWER,       (uint8_t []){0x48, 0x12, 0x72, 0x09, 0x32, 0x54, 0x71, 0x71, 0x57, 0x47}, 10,0},
    {UD_SETDISP,        (uint8_t []){0x00, 0x80, 0x64, 0x0C, 0x0D, 0x2F}, 6,0},
    {UD_SETCYC,         (uint8_t []){0x73, 0x74, 0x73, 0x74, 0x73, 0x74, 0x01, 0x0C, 0x86, 0x75, 0x00, 0x3F, 0x73, 0x74, 0x73, 0x74, 0x73, 0x74, 0x01, 0x0C, 0x86}, 21,0},
    {UD_SETGIP0,        (uint8_t []){0x00, 0x00, 0x07, 0x07, 0x40, 0x07, 0x0C, 0x00, 0x08, 0x10, 0x08, 0x00, 0x08, 0x54, 0x15, 0x0A, 0x05, 0x0A, 0x02, 0x15, 0x06, 0x05, 0x06, 0x47, 0x44, 0x0A, 0x0A, 0x4B, 0x10, 0x07, 0x07, 0x0C, 0x40}, 33,0},
    {UD_SETGIP1,        (uint8_t []){0x1C, 0x1C, 0x1D, 0x1D, 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x24, 0x25, 0x18, 0x18, 0x26, 0x27, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x20, 0x21, 0x18, 0x18, 0x18, 0x18}, 44,0},
    {UD_SETGIP2,        (uint8_t []){0x1C, 0x1C, 0x1D, 0x1D, 0x07, 0x06, 0x05, 0x04, 0x03, 0x02, 0x01, 0x00, 0x0B, 0x0A, 0x09, 0x08, 0x21, 0x20, 0x18, 0x18, 0x27, 0x26, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x18, 0x25, 0x24, 0x18, 0x18, 0x18, 0x18}, 44,0},
    {UD_SETVCOM,        (uint8_t []){0x6E, 0x6E}, 2,0}, //caiji
    {UD_SETGAMMA,       (uint8_t []){0x00, 0x0A, 0x15, 0x1B, 0x1E, 0x21, 0x24, 0x22, 0x47, 0x56, 0x65, 0x66, 0x6E, 0x82, 0x88, 0x8B, 0x9A, 0x9D, 0x98, 0xA8, 0xB9, 0x5D, 0x5C, 0x61, 0x66, 0x6A, 0x6F, 0x7F, 0x7F, 0x00, 0x0A, 0x15, 0x1B, 0x1E, 0x21, 0x24, 0x22, 0x47, 0x56, 0x65, 0x65, 0x6E, 0x81, 0x87, 0x8B, 0x98, 0x9D, 0x99, 0xA8, 0xBA, 0x5D, 0x5D, 0x62, 0x67, 0x6B, 0x72, 0x7F, 0x7F}, 58,0},
    {0xC0,              (uint8_t []){0x1F, 0x31}, 2,0},
    {UD_SETPANEL,       (uint8_t []){0x03}, 1,0},
    {0xD4,              (uint8_t []){0x02}, 1,0},
    {UD_SETREGBANK,     (uint8_t []){0x02}, 1,0},
    {UD_SETGIP3,        (uint8_t []){0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF}, 12,0},
    {UD_SETREGBANK,     (uint8_t []){0x00}, 1,0},
    {UD_SETREGBANK,     (uint8_t []){0x01}, 1,0},
    {UD_SETPOWER,       (uint8_t []){0x00}, 1,0},
    {UD_SETREGBANK,     (uint8_t []){0x00}, 1,0},
    {0xBF,              (uint8_t []){0x40, 0x81, 0x50, 0x00, 0x1A, 0xFC, 0x01}, 7,0},
    {0xC6,              (uint8_t []){0xED}, 1,0},
	{0x11, (uint8_t[]){0x00}, 0, 120},
    {0x29, (uint8_t[]){0x00}, 0, 0},
};

moononournation, 2025-04-17:
I do not have this display in hand, but it is welcome you contribute a pull request

Big-mouth-c, 2025-04-18:
我拉取了一个 跟新请求，应该符合github约定。国内访问github  太难了，页面时常打不开，使用 Watt Toolkit 加速又无法上传或下载文件。

closed 2025-04-25 by moononournation