moononournation/Arduino_GFX #664: ZX3D50CE02S or "WT32-SC01 PLUS" does not compile

knopserl, 2025-04-21:
I used the example 
PDQgraphicstest.ino

I use the well known display WT32-SC01 PLUS (ZX3D50CE02S) which results in the Arduino_DataBus *bus = new Arduino_ESP32LCD8.
but this is not defined

#define GFX_DEV_DEVICE ZX3D50CE02S
#define GFX_BL 45
Arduino_DataBus *bus = new Arduino_ESP32LCD8(
    0 /* DC */, GFX_NOT_DEFINED /* CS */, 47 /* WR */, GFX_NOT_DEFINED /* RD */,
    9 /* D0 */, 46 /* D1 */, 3 /* D2 */, 8 /* D3 */, 18 /* D4 */, 17 /* D5 */, 16 /* D6 */, 15 /* D7 */);
Arduino_GFX *gfx = new Arduino_ST7796(bus, 4 /* RST */, TFT_rot /* rotation */, true /* IPS */);

#if defined(ESP32) && (CONFIG_IDF_TARGET_ESP32S3)
this preprocessor condition is not met!
#if (!defined(ESP_ARDUINO_VERSION_MAJOR)) || (ESP_ARDUINO_VERSION_MAJOR < 3)

thus Arduino_ESP32LCD8 is not defined 
compiler error:
Arduino_GFX_dev_device.h:796:28: error: expected type-specifier before 'Arduino_ESP32LCD8'

So this display cant be used. I also tried Arduino_ESP32PAR8 but this does also not work.

I use the Arduino IDE 2.3.6, and your latest Arduino_GFX library. I've tried the latest ESP32 API (board 3.2 = IDF 5.4 and 3.0.7 = IDF 5.1)

Any hints?

Big-mouth-c, 2025-04-23:
Use Arduino-ESP32 version 2.x

knopserl, 2025-04-25:
> Use Arduino-ESP32 version 2.x

not a good advice, that will cause a lot of other conflicts in my projects.
I'd rather would like to see support for v3.2 and LVGL 9.2 than going backwards

moononournation, 2025-04-26:
The main branch updated Arduino_ESP32LCD8, you may try it again.

closed 2025-04-26 by moononournation