moononournation/Arduino_GFX #669: ZX3D50CE02S or "WT32-SC01 PLUS" does not compile

knopserl, 2025-04-30:
> The main branch updated Arduino_ESP32LCD8, you may try it again. 
 _Originally posted by @moononournation in [#664](https://github.com/moononournation/Arduino_GFX/issues/664#issuecomment-2831658298)_

This issue has been closed but seem not be resolved. Still a lot of compile errors in the described environment in the original post

several compile errors in the file: GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp

d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'virtual bool Arduino_ESP32QSPI::begin(int32_t, int8_t)':
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:62:21: error: 'ESP_INTR_CPU_AFFINITY_AUTO' was not declared in this scope
   62 |       .isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO,
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32LCD8.cpp: In member function 'virtual bool Arduino_ESP32LCD8::begin(int32_t, int8_t)':
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32LCD8.cpp:791:29: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
  791 |       .max_transfer_bytes = LCD_MAX_PIXELS_AT_ONCE * 2,
      |                             ^~~~~~~~~~~~~~~~~~~~~~
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32LCD8.cpp:824:24: error: 'esp_lcd_i80_alloc_draw_buffer' was not declared in this scope
  824 |   _buffer = (uint8_t *)esp_lcd_i80_alloc_draw_buffer(_io_handle, LCD_MAX_PIXELS_AT_ONCE * 2 + 16, MALLOC_CAP_DMA);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32LCD8.cpp: In member function 'virtual void Arduino_ESP32LCD8::write(uint8_t)':
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32LCD8.cpp:894:21: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
  894 |   if (_bufferLen >= LCD_MAX_PIXELS_AT_ONCE * 2)
      |                     ^~~~~~~~~~~~~~~~~~~~~~
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32LCD8.cpp: In member function 'virtual void Arduino_ESP32LCD8::write16(uint16_t)':
d:\data\mcdev\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32LCD8.cpp:910:21: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
  910 |   if (_bufferLen >= LCD_MAX_PIXELS_AT_ONCE * 2 - 4)
      |                     ^~~~~~~~~~~~~~~~~~~~~~
exit status 1

This error has also been reported by others in Jan 2025 and I have not seen, that it was resolved.

eggfly, 2025-05-02:
+1
And tag=v1.5.6 compile OK!
```

.pio/libdeps/default/GFX Library for Arduino/src/databus/Arduino_ESP32LCD8.cpp: In member function 'virtual bool Arduino_ESP32LCD8::begin(int32_t, int8_t)':
.pio/libdeps/default/GFX Library for Arduino/src/databus/Arduino_ESP32LCD8.cpp:791:29: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
  791 |       .max_transfer_bytes = LCD_MAX_PIXELS_AT_ONCE * 2,
      |                             ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/default/GFX Library for Arduino/src/databus/Arduino_ESP32LCD8.cpp:801:18: warning: narrowing conversion of '((Arduino_ESP32LCD8*)this)->Arduino_ESP32LCD8::<anonymous>.Arduino_DataBus::_speed' from 'int32_t' {aka 'long int'} to 'uint32_t' {aka 'long unsigned int'} [-Wnarrowing]
  801 |       .pclk_hz = _speed,
      |                  ^~~~~~
.pio/libdeps/default/GFX Library for Arduino/src/databus/Arduino_ESP32LCD8.cpp: In member function 'virtual void Arduino_ESP32LCD8::write(uint8_t)':
.pio/libdeps/default/GFX Library for Arduino/src/databus/Arduino_ESP32LCD8.cpp:894:21: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
  894 |   if (_bufferLen >= LCD_MAX_PIXELS_AT_ONCE * 2)
      |                     ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/default/GFX Library for Arduino/src/databus/Arduino_ESP32LCD8.cpp: In member function 'virtual void Arduino_ESP32LCD8::write16(uint16_t)':
.pio/libdeps/default/GFX Library for Arduino/src/databus/Arduino_ESP32LCD8.cpp:910:21: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
  910 |   if (_bufferLen >= LCD_MAX_PIXELS_AT_ONCE * 2 - 4)
      |                     ^~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/default/lib97d/GFX Library for Arduino/databus/Arduino_ESP32LCD8.cpp.o] Error 1
```

wapel, 2025-05-03:
+1
Arduino IDE 2.3.5

```
Arduino_ESP32LCD8.cpp:791:29: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
  791 |       .max_transfer_bytes = LCD_MAX_PIXELS_AT_ONCE * 2,
```

knopserl, 2025-05-03:
> +1 Arduino IDE 2.3.5
> 
> ```
> Arduino_ESP32LCD8.cpp:791:29: error: 'LCD_MAX_PIXELS_AT_ONCE' was not declared in this scope
>   791 |       .max_transfer_bytes = LCD_MAX_PIXELS_AT_ONCE * 2,
> ```

I get this undecalred LCD_MAX_PIXELS_AT_ONCE compile error but also the undeclared ESP_INTR_CPU_AFFINITY_AUTO with the also the same/latest Arduino IDE.
I also have to mention, that I still use arduino-esp32 3.0.7 ) ESP API 5.1.x and not 3.2 because I get then runtime errors in AsyncTCP and ESPAsyncWebServer which is related to chances made after ESPRESSIF ESP32 API  5.1.
But I cant also not switch back to arduino-esp32 2.x (way to old and causing other conflicts)

closed 2025-05-04 by moononournation