moononournation/Arduino_GFX #687: new_table() malloc: 32776

silardg, 2025-06-19:
The issue is the following:
- I have an SD Card and display connect to an esp32-s3
- The Gif is working and running but is restarted after the error shows up

Tried:
- different malloc areas - PSRAM etc

I have switched to using gif as 240x960x2 is too large for the memory and can't use mjpeg. 

Configuration for the arduino IDE is the following:
[image]

The gif is being interrupted by this error:
`ERROR: gd_get_frame() failed!
GIF video end
Actual duration: 4479, expected duration: 4400, remain: 46 (1.05%)
new_table() malloc: 32776
GIF video start
unknown extension: 40
Read sep: [f].`

I'm using the qualia esp32-s3 rgb 666 with https://www.adafruit.com/product/5799

I have also optimised the gif and made the screen look much better through this command:
`ffmpeg -i BLOK_Animation_V1_1-ezgif.com-rotate.gif -vf "scale=240:960,fps=10" -c:v gif -y optimized.gif`

My whole code:

```
#define GIF_FILENAME "/video.gif"
//ffmpeg -i BLOK_Animation_V1_1-ezgif.com-rotate.gif -vf "scale=240:960,fps=10" -c:v gif -y optimized.gif
#include <Arduino_GFX_Library.h>
#include <SD_MMC.h>

#include "GifClass.h"
static GifClass gifClass;
Arduino_XCA9554SWSPI *expander = new Arduino_XCA9554SWSPI(
    PCA_TFT_RESET, PCA_TFT_CS, PCA_TFT_SCK, PCA_TFT_MOSI, &Wire, 0x3F);

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    TFT_DE, TFT_VSYNC, TFT_HSYNC, TFT_PCLK,
    TFT_R1, TFT_R2, TFT_R3, TFT_R4, TFT_R5,
    TFT_G0, TFT_G1, TFT_G2, TFT_G3, TFT_G4, TFT_G5,
    TFT_B1, TFT_B2, TFT_B3, TFT_B4, TFT_B5,
    1 /* hsync_polarity */, 20 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 20 /* hsync_back_porch */,
    1 /* vsync_polarity */, 20 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */,
    0 /* pclk_active_neg */, GFX_NOT_DEFINED /* prefer_speed */, false /* useBigEndian */, 0 /* de_idle_high */,
    0 /* pclk_idle_high */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    240 /* width */, 960 /* height */, rgbpanel, 0 /* rotation */, false /* auto_flush */,
    expander, GFX_NOT_DEFINED /* RST */, HD371001C40_init_operations, sizeof(HD371001C40_init_operations), 120 /* col_offset1 */);

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println("MJPEG Video Playback Demo");

#ifdef GFX_EXTRA_PRE_INIT
  GFX_EXTRA_PRE_INIT();
#endif

  // Initialize display
  Wire.setClock(400000); // Speed up I2C
  if (!gfx->begin()) {
    Serial.println(F("Display initialization failed!"));
    while (1) delay(100);
  }

  // Test display with a color pattern
  gfx->fillScreen(BLACK);
  gfx->drawRect(0, 0, 240, 960, WHITE); // Full-screen border
  gfx->fillRect(0, 0, 240, 320, RED);   // Top third (red)
  gfx->fillRect(0, 320, 240, 320, GREEN); // Middle third (green)
  gfx->fillRect(0, 640, 240, 320, BLUE); // Bottom third (blue)
  Serial.println(F("Displaying RGB color pattern for 5 seconds..."));
  delay(500);
  gfx->fillScreen(BLACK);

  // Initialize SD card
  SD_MMC.setPins(SCK, MOSI, MISO, A0, A1, SS); // Quad MMC
  // if (!SD_MMC.begin("/root", false)) {
  if(!SD_MMC.begin("/root", false, false, SDMMC_FREQ_HIGHSPEED, 16)) {
    Serial.println(F("ERROR: File System Mount Failed!"));
    gfx->println(F("ERROR: File System Mount Failed!"));
    while (1) delay(100);
  }
  Serial.println(F("Found SD Card"));

  // Configure buttons
  expander->pinMode(PCA_BUTTON_UP, INPUT);
  expander->pinMode(PCA_BUTTON_DOWN, INPUT);
}

void loop() {

  File gifFile = SD_MMC.open(GIF_FILENAME, "r");

  if (!gifFile || gifFile.isDirectory())
  {
    Serial.println(F("ERROR: open gifFile Failed!"));
    gfx->println(F("ERROR: open gifFile Failed!"));
  }
  else
  {
    // read GIF file header
    gd_GIF *gif = gifClass.gd_open_gif(&gifFile);
    if (!gif)
    {
      Serial.println(F("gd_open_gif() failed!"));
    }
    else
    {
      uint8_t *buf = (uint8_t *)malloc(gif->width * gif->height);
      if (!buf)
      {
        Serial.println(F("buf malloc failed!"));
      }
      else
      {
        int16_t x = (gfx->width() - gif->width) / 2;
        int16_t y = (gfx->height() - gif->height) / 2;

        Serial.println(F("GIF video start"));
        int32_t start_ms = millis(), t_delay = 0, delay_until;
        int32_t res = 1;
        int32_t duration = 0, remain = 0;
        while (res > 0)
        {
          t_delay = gif->gce.delay * 10;
          res = gifClass.gd_get_frame(gif, buf);
          if (res < 0)
          {
            Serial.println(F("ERROR: gd_get_frame() failed!"));
            break;
          }
          else if (res > 0)
          {
            gfx->drawIndexedBitmap(x, y, buf, gif->palette->colors, gif->width, gif->height);

            duration += t_delay;
            delay_until = start_ms + duration;
            while (millis() < delay_until)
            {
              delay(1);
              remain++;
            }
          }
        }
        Serial.println(F("GIF video end"));
        Serial.print(F("Actual duration: "));
        Serial.print(millis() - start_ms);
        Serial.print(F(", expected duration: "));
        Serial.print(duration);
        Serial.print(F(", remain: "));
        Serial.print(remain);
        Serial.print(F(" ("));
        Serial.print(100.0 * remain / duration);
        Serial.println(F("%)"));

        gifClass.gd_close_gif(gif);
        free(buf);
      }
    }
  }
}
```

silardg, 2025-06-19:
```
Free heap: 317160 bytes
Free PSRAM: 7350456 bytes
Free heap: 317160 bytes
Free PSRAM: 7350456 bytes
unknown extension: 40
Read sep: [f].

ERROR: gd_get_frame() failed!
GIF video end
Actual duration: 4479, expected duration: 4400, remain: 44 (1.00%)
new_table() malloc: 32776
GIF video start
Free heap: 317160 bytes
Free PSRAM: 7350456 bytes
Free heap: 317160 bytes
Free PSRAM: 7350456 bytes
```

Added heap and psram reading... couldn't find anything yet

moononournation, 2025-06-19:
memory allocation is always a big topic in programming