moononournation/Arduino_GFX #698: Help of NV3007 driver issue

qqilin1213, 2025-07-30:
I created the initial file to drive the NV3007 (142x428) with the ESP32C3, but it does not work.

In 金逸晨_nv3007_success, I can display different screen colors but cannot display text; in helloWorld_error, neither the background nor text can be displayed.

[金逸晨_nv3007_success.txt](https://github.com/user-attachments/files/21510362/_nv3007_success.txt)
[helloWorld_error.txt](https://github.com/user-attachments/files/21510363/helloWorld_error.txt)

[](https://github.com/qqilin1213/picx-images-hosting/raw/master/20250730/link.9kghkj9a0q.webp)

[image]

qqilin1213, 2025-07-30:
[image]

Gazyli, 2025-07-31:
Hey, I am working on an alternate driver, with a different initialization sequence for the NV3007, as the current implementation does not correctly work with my display as well:

[image]

I will be making a pull request with the alternate driver later today, hope it will fix your issues as well.

closed 2025-08-01 by qqilin1213