moononournation/Arduino_GFX #708: CO5300 driver issue

legregz, 2025-08-07:
I'm coding a esp32-S3 on a waveshare product ( [](https://www.waveshare.com/esp32-s3-touch-amoled-1.75.htm) ) with a CO5300 display driver but there is a lot of bugs : all draw methods didn't work except for the fillRect() method and only for dimensions > 1. I can't do
`gfx->fillRect(203, 203, 1, 60, RED)` or `gfx->fillRect(203, 203, 60, 1, RED)` for exemple. The `drawCircle()`, `drawRoundRect()` draws strange things with black lines in the "circle". The `gfx->drawFastHLine()` and the `gfx->drawFastVLine()` didn't works.

moononournation, 2025-08-07:
If PDQgraphicstest example using canvas, please use canvas

legregz, 2025-08-08:
I didn't use PQD_GFX but directly Arduino_GFX. I tried with a canvas but this didn't work anymore. Am i supposed to use PQD instead of Arduino_GFX ? 

PS : I'm new to esp32 so there are a lot of things I don't understand.

-- Sent from /e/OS Mail.

moononournation, 2025-08-10:
Please try PDQgraphicstest.ino example in Arduino_GFX first.

closed 2025-08-10 by moononournation