moononournation/Arduino_GFX #717: Flickering since version 1.6.1

F1ATB, 2025-08-29:
Hello

I used with satisfaction this library for a module [JC3248W535](https://s.click.aliexpress.com/e/_oExVwIp)
I have a simple example here [https://f1atb.fr/esp32-s3-3-5-inch-capacitive-touch-ips-display-setup/](https://f1atb.fr/esp32-s3-3-5-inch-capacitive-touch-ips-display-setup/)
It runs perfectly with library version 1.6.0. With library 1.6.1 I have some flickering. I did the test with 2 modules.

Best Regards and thank you for this development.

André / F1ATB
https://f1atb.fr/

Flo082002, 2025-09-05:
```
#define LCD_QSPI_CS   12
#define LCD_QSPI_CLK  5
#define LCD_QSPI_D0   1
#define LCD_QSPI_D1   2
#define LCD_QSPI_D2   3
#define LCD_QSPI_D3   4

#define I2C_SDA       8
#define I2C_SCL       7
#define GFX_BL        6

#define SCREEN_WIDTH  320
#define SCREEN_HEIGHT 480

Arduino_DataBus *bus = new Arduino_ESP32QSPI(LCD_QSPI_CS, LCD_QSPI_CLK, LCD_QSPI_D0, LCD_QSPI_D1, LCD_QSPI_D2, LCD_QSPI_D3);
Arduino_AXS15231B *g = new Arduino_AXS15231B(bus, GFX_NOT_DEFINED, 0, false, SCREEN_WIDTH, SCREEN_HEIGHT);
Arduino_Canvas *gfx  = new Arduino_Canvas(SCREEN_WIDTH, SCREEN_HEIGHT, g, 0, 0, 0);
```

moononournation, 2025-09-13:
Should be fixed. Please get update and try:
```
#define GFX_BL 1
Arduino_DataBus *bus = new Arduino_ESP32QSPI(
    45 /* CS */, 47 /* SCK */, 21 /* D0 */, 48 /* D1 */, 40 /* D2 */, 39 /* D3 */);
Arduino_GFX *g = new Arduino_AXS15231B(
    bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */, false /* IPS */, 320 /* width */, 480 /* height */,
    0 /* col offset 1 */, 0 /* row offset 1 */, 0 /* col offset 2 */, 0 /* row offset 2 */,
    axs15231b_320480_init_operations, sizeof(axs15231b_320480_init_operations));
#define CANVAS
Arduino_Canvas *gfx = new Arduino_Canvas(
    320 /* width */, 480 /* height */, g, 0 /* output_x */, 0 /* output_y */, 0 /* rotation */);
```

closed 2025-09-13 by moononournation

F1ATB, 2025-09-13:
I tested your suggestion and Arduino IDE compiler return 👍 
Compilation error: 'axs15231b_320480_init_operations' was not declared in this scope; did you mean 'axs15231b_init_operations

What to do ?
Regards
André

moononournation, 2025-12-27:
Now separated 2 type of init operation sequence, axs15231b_320480_type1_init_operations and axs15231b_320480_type2_init_operations.