moononournation/Arduino_GFX #724: Crash when WiFi.scanNetworks() is called. ESP32-S3-Dev-Kit-1 R16N8

axellottel-jpg, 2025-09-25:
I have an ESP32-S3-Dev-Kit-1 module, a 2.4" ST7789 display.
It has 16M Flash and 8M PSRAM,
I can display text and graphics on the display.
When I compile the WiFi analyzer the ESP reboots when the WiFi.scanNetworks() is called.
If I put the WiFi.scanNetworks() before the display is initialized the sketch runs but in the example you only get the scan on boot-up.
I have also tested with the 1.8" ST7735 display and get the same results.

Here is the definition code:
```
#include <Arduino_GFX_Library.h>
#define TFT_CS 40 // GFX_NOT_DEFINED for display without CS pin
#define TFT_DC 41
#define TFT_RST 42
#define GFX_BL 48
Arduino_DataBus *bus = new Arduino_ESP32SPI(TFT_DC, TFT_CS, 36 /* SCK */, 35 /* MOSI */, GFX_NOT_DEFINED /* MISO */, HSPI /* spi_num */);
Arduino_GFX *gfx = new Arduino_ST7789(bus, TFT_RST, 0 /* rotation */, false /* IPS */, 240 /* width */, 320 /* height */, 0 /* col offset 1 */,0 /* row offset 1 */, 2 /* col offset 2 */, 1 /* row offset 2 *//*, false  BGR */);

```
Debug Dump:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1050
load:0x403c8700,len:0xaf4
load:0x403cb700,len:0x2e7c
entry 0x403c8898
[     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42013370
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4201333c
[    24][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42013308
[    35][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x420132d4
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42013370
[    58][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4201333c
[    69][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42013308
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x420132d4
[    92][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42013370
[   103][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4201333c
[   114][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42013308
[   126][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x420132d4
[   169][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc97dd0
[   180][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc97dd0
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   352668 B ( 344.4 KB)
  Free Bytes        :   315084 B ( 307.7 KB)
  Allocated Bytes   :    32552 B (  31.8 KB)
  Minimum Free Bytes:   309816 B ( 302.6 KB)
  Largest Free Block:   258036 B ( 252.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
[... 72 lines cut ...]
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8374084 B (8177.8 KB)
  Allocated Bytes   :    11620 B (  11.3 KB)
  Minimum Free Bytes:  8373876 B (8177.6 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    35 : SPI_MASTER_MOSI[1]
    36 : SPI_MASTER_SCK[1]
    40 : GPIO
    41 : GPIO
    42 : GPIO
    43 : UART_TX[0]
    44 : UART_RX[0]
    48 : GPIO
============ After Setup End =============

```
The only lines changed are the above so I would guess that the example should work.

moononournation, 2025-09-25:
I suggest you try PDQgraphicstest example first.

axellottel-jpg, 2025-09-26:
It works fine (PDQgraphicstest). Several other examples work but when I use anyhting that uses WiFi.h I run into troubles, mainly the WiFi.scanNetworks().
I am in the process of adding WiFiManager as it does a lot with WiFI including scan for networks.

closed 2025-10-19 by moononournation